ArduCAM/Arduino #164: OV5642 Transmitting Images over XBEE

jwerbs1217, 2017-04-04:
Hi,

I am attempting to send images over an XBEE, but I believe it uses serial whereas the normal setup with your given arducam_host.exe uses SPI through the usb cord. I have an OV5642 camera shield hooked up directly to an arduino. How can I get the image to be compressed to be sent through serial through the XBEE and be opened on the other side? 

Thanks!

supprot, 2017-04-04:
@jwerbs1217 
Hi, 
Yes, In our demo, we send the JPEG image datas to arducam_host.exe by serial port.
If you want to reveive the data .the head of data is 0xff 9xd8 and the end of data is 0xff 0xd9.
let us know if you need more help.
Regards,
ArduCAM support team.

jwerbs1217, 2017-04-04:
Is there a simple way to change the low power code to make it take like 30 photos in a row and send them as individual packets?

jwerbs1217, 2017-04-14:
I figured out how to make it repeat and I believe I have it sending an image to serial, but how do I interpret it as an image? Is there source code for that part of the arducam_host that I can see? I just need to take the stream of data from 0xff 9xd8 to 0xff 0xd9, but how do I do that, and how do I display that as an image? Thank you

supprot, 2017-04-15:
@jwerbs1217 
Hi,

Very sorry for we don't supply the source of our arducam_Host.
Our Host support single and streamimg mode.

Let us know if you need more help.
Regards,
ArduCAM support team.

jwerbs1217, 2017-04-18:
is there any information you can tell me about the stream of data that will help me render this as an image myself? I just now am able to see all of it but want to write a python or MATLAB script to handle it all and just display the image on screen, as for my project I am unable to use your host software

jwerbs1217, 2017-04-19:
My data seems to only be ~18000 bytes long whereas I am trying to take a 640x480 image. Shouldn't this have 640*480*3(=921600) bytes? How can I analyze this?

supprot, 2017-04-19:
@jwerbs1217 
In our demo ,the image data has been compressed in JPEG format.
So the image datas you get is not the original datas.
You can configure it out put in RGB format without compression.
Please refer to this example.
// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules.
//This demo can only work on ARDUCAM_SHIELD_V2 platform.
//This demo is compatible with ESP8266
// This demo was made for Omnivision OV2640/OV5640/OV5642/ sensor.
// It will turn the ArduCAM into a real digital camera with capture and playback functions.
// 1. Preview the live video on LCD Screen.
// 2. Capture and buffer the image to FIFO when shutter pressed quickly.
// 3. Store the image to Micro SD/TF card with BMP format.
// 4. Playback the capture photos one by one when shutter buttom hold on for 3 seconds.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM shield V2
// and use Arduino IDE 1.6.8 compiler or above
#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo was made for Omnivision MT9D111A/MT9D111B/MT9M112/MT9V111_CAM/
//                                  MT9M001/MT9T112/MT9D112/OV7670/OV7675/
//                                  OV7725/OV2640/OV3640/OV5640/OV5642 sensor.
#if !(defined ARDUCAM_SHIELD_V2 && (defined MT9D111A_CAM|| defined MT9D111B_CAM || defined MT9M112_CAM  \
                                 || defined MT9V111_CAM || defined MT9M001_CAM || defined MT9T112_CAM \
                                 || defined MT9D112_CAM || defined OV7670_CAM  || defined OV7675_CAM  \
                                 || defined OV7725_CAM  || defined OV2640_CAM  || defined OV5640_CAM  \
                                 || defined OV5642_CAM|| defined OV3640_CAM))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#if defined(__arm__)
  #include <itoa.h>
#endif
#if defined(ESP8266)
 #define SD_CS 0 
 const int SPI_CS = 16;
#else 
 #define SD_CS 9 
 const int SPI_CS =10;
#endif
#define BMPIMAGEOFFSET 66
const int bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
#if defined (MT9D111A_CAM)
  ArduCAM myCAM(MT9D111_A, SPI_CS);
#elif defined (MT9D111B_CAM)
  ArduCAM myCAM(MT9D111_B, SPI_CS);
#elif defined (MT9M112_CAM)
  ArduCAM myCAM(MT9M112, SPI_CS);
#elif defined (MT9V111_CAM)
  ArduCAM myCAM(MT9V111, SPI_CS);
#elif defined (MT9M001_CAM)
  ArduCAM myCAM(MT9M001, SPI_CS);
#elif defined (MT9T112_CAM)
  ArduCAM myCAM(MT9T112, SPI_CS);
#elif defined (MT9D112_CAM)
  ArduCAM myCAM(MT9D112, SPI_CS);
#elif defined (OV7670_CAM)
  ArduCAM myCAM(OV7670, SPI_CS);
[... 221 lines cut ...]
//Only support RGB565 bmp format
void dispBitmap(File inFile)
{ 
  char VH = 0, VL = 0;
  int i, j = 0;
  for (i = 0 ; i < BMPIMAGEOFFSET; i++)
  inFile.read();
  for (i = 0; i < 320; i++)
  for (j = 0; j < 240; j++)
  {
    VL = inFile.read();
    //Serial.write(VL);
    VH = inFile.read();
    //Serial.write(VH);
    #if defined(ESP8266)
      yield();
    #endif
    myGLCD.LCD_Write_DATA(VH, VL);
  }
  myGLCD.clrXY();
}

Let us know if you need more help.
Regards,
ArduCAM support team.

jwerbs1217, 2017-04-20:
Thank you! That is very helpful info

ArduCAM, 2018-08-13:
You have to follow our Arduino example and port to your Xbee module by yourself.

Badhr, 2018-08-13:
Ok, would please send me the link of your Arduino example.
Thank you,

danishmanzoor757, 2019-05-30:
@jwerbs1217  did you gain success in doing so?
kindly guide me through