ArduCAM/Arduino #166: OV5642 image output is torn and blocky

NVergunst, 2017-04-07:
Hello,

I am using the ArduCAM 5MP (Rev B) and I am having terrible trouble getting anything useful out of it. I can talk to it via I2C, and talk to it via SPI. I can start captures, read the FIFO length and clock out data. I can see it all on an oscilloscope and it looks great. When I reassemble the bytes, I just get garbage though. Anyone familiar with the issue, seen it before or anything? 

I am not using an Arduino, but writing standard C on a PIC32. Everything else is working. The full app is reading chunks of data from the SPI bus and transferring via UART to a cellular transceiver which is FTP-ing the data to a server. I can transfer random data chunks with known data from the micro and receive the exact same (correct) bytes on the FTP server. So I am inclined to believe that the problem is somewhere in the ArduCAM. Either is setup or execution. 

Any help is appreciated, this one part has blown out my schedule and I am on the brink of dropping it and getting a new module...

[image]
[image]

NVergunst, 2017-04-07:
I see in another issue ticket with a rely from April 20th, 2016 where it was said:

> @Juan
> There is SPI timing error with 5MP when used in ESP8266, please try our special library released for ESP8266 from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM

"SPI Timing Error" is interesting... Is the waveform from the datasheet incorrect? Is there more of a minimum hold time before clocking than lead to believe? I am running around 2Mhz, which I thought should be well below the 8MHz rate the datasheet says it is capable of.

NVergunst, 2017-04-08:
Looks like the special timing thing in the ESP code is the "yield" arduino function to keep the background tasks going. That's obviously not my problem.

Is there any way for me to write a pattern to the FIFO and then read the pattern back to see if it is skipping or something???

supprot, 2017-04-08:
@NVergunst 
Hi,
Please try to narrow the data buffer?
Let us know if you need more help。
Regards,
ArduCAM support team.

NVergunst, 2017-04-08:
Narrow the buffer on which end? 

I have tried with reading 1K at a time, 100bytes at a time, and even 1 byte at a time. It seems like if I read 100-250 bytes at a time, I get the most image out. If I read 1 byte at a time, the image doesn't even become a jpeg. Reading 1K at a time, I get more streaks than usual.

NVergunst, 2017-04-08:
Is there a way for me to read the FIFO pointer from the ArduCAM? Make sure it is being set to index + bytes read and/or set it back and re-read data to verify integrity?

NVergunst, 2017-04-08:
[image]
[image]
[image]

supprot, 2017-04-08:
@NVergunst  
Hi,
Our camera support point reset.
You can refer to the below code.
uint8_t read_fifo_again(ArduCAM myCAM)
{  
  uint8_t temp, temp_last;
  uint32_t length = 0;
  temp = myCAM.read_reg(ARDUCHIP_FIFO);
  myCAM.write_reg(ARDUCHIP_FIFO, temp|FIFO_RDPTR_RST_MASK);  //FIFO_RDPTR_RST_MASK
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= FIFO_SIZE) //8M
  {
    Serial.println("Over size.");
    return 0;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println("Size is 0.");
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  SPI.transfer(0x00);//First byte is 0xC0 ,not 0xff
  length--;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
   if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(12);
  }
  myCAM.CS_HIGH();
  is_header = false;
}

Let us know if you need more help.
Regards,
ArduCAM support team.

NVergunst, 2017-04-08:
Is there a minimum time between back-to-back burst reads required? I'm not understanding why the image data is time variant. If I go slowly, I get more data out. But at speed, the signals look excellent and read back properly. It is just seems to skip. I will try to implement the SPI read of the FIFO address. Thanks for the quick support so far.

supprot, 2017-04-08:
@NVergunst 
Hi,
Please send your code to us and we will test it for you.
What's the speed of your SPI speed?
Let us know if you need more help.
Regards,
ArduCAM support team.

NVergunst, 2017-04-08:
SPI Speed is roughly 2MHz. But I have slowed it down to 400KHz without change.

[image]

NVergunst, 2017-04-08:
My code is written for a Microchip PIC32MX440F512H on a custom board. I doubt it will do much good. It's actually based off of the code here on github but all the routines were updated to work with the PIC peripherals and time slicing.

NVergunst, 2017-04-08:
I just got my first real pictures out of the thing. I kept the SPI speed at 2MHz, but inserted 100ms delays between each 100bytes read. This gave me a full image out. Really really slowly, but it worked. So the problem is most definitely timing related...

supprot, 2017-04-08:
@NVergunst 
Hi,
Maybe it is related to the platform of the PIC32.The SPI 
should not be configured well or have conflicts with other functions.
We have not tested it on the PIC32 platform. 
Regards,
ArduCAM support team.

NVergunst, 2017-04-08:
> @NVergunst
> Hi,
> Maybe it is related to the platform of the PIC32.The SPI
> should not be configured well or have conflicts with other functions.
> We have not tested it on the PIC32 platform.
> Regards,
> ArduCAM support team.

Nope, the SPI is working perfectly. I am running a single thread operation with only 2 interrupts (I2C and UART). The way the program is made, the I2C interrupt is silent when the SPI is getting data since the master is the PIC32 as well. The UART could get interrupted at any time (likelyhood low) but the SPI doesn't care, it would just stretch a pulse. I've had this same type of code running at 40MHz+, I know it is not the problem.

NVergunst, 2017-04-08:
What's the proper way of polling the device to know that the capture is complete and ready for readback from the FIFO? If I wait multiple seconds between reads, by inserting a breakpoint, it seems to transfer and image every time now. But if I let it free-run, I get garbage.

NVergunst, 2017-04-08:
And any idea about the scan-lines in the captured image?

[image]

DallonTG, 2017-05-01:
I have been having similar issues for months now. I have noticed that even when I reset the FIFO read pointer and read the same image multiple times, the data is different (although the number of bytes I receive is exactly the same). I can never get a good image from the Arducam Mini.

Inserting delays between reads doesn't seem to help in my case. Has anyone learned what the issue is more? I am fairly certain it is a timing issue on the Arudcam's side.

NVergunst, 2017-05-01:
Since I have no "official" answer, in my case reading slowly with delays between transactions, gives me a workable image most of the time. In my tests, it's about 1 corrupt image in 350. This only applies to VGA quality images though, where the output JPG is roughly 80KB in size. If I capture a 5MP image, I can read out the data and most of the time it is corrupt. A full proper image is only uploaded maybe 1 in 90 times. 

I am inclined to believe it is something speed/buffer related since the beginning of the image always looks correct. Always. Then as the image progresses it gets worse and worse. With the JPG encoding, it looks like basically it is missing information and then the rest of the image gets filled in.

I know my SPI code is rock solid. I have been using these things for over a decade and they are in production quality hardware. I am not a hobbyist playing with an Arduino, I'm a full EE/CE that has been doing this professionally for some time now. 

And I solved my line problem above, you have to enable the line filtering in the OV module itself over the I2C. That datasheet is huge, but it has most of the info. The built in libraries supplied for the Arducam are pretty silly and just don't work for me. Maybe something magically in Arduino land happens to make it work, but in real embedded C land, they don't work. Only thing I didn't find in the OV datasheet or in the Arducam software is for auto-focus, as they say contact an apps engineer to find out how. My guess is there's an errata there and they don't want to publish it so they give out the information on a piece by piece basis....

DallonTG, 2017-05-03:
Thanks for your fast response! 

I am also fairly certain that my SPI code and my I2C code is correct. I am also not using an Arduino. I'm doing this in bare metal with a SmartFusion2. 

I was finally able to get a clean image from the Arducam. I had to insert a 20ms delay between every single byte that I read and send over UART. Because of this, it literally takes more than an hour to get an image. This speed is extremely slow. I need to be saving images within seconds.

So there is definitely a timing problem with reading the image from the buffer. @supprot @ArduCAM How long does it take for the next byte to be in the read FIFO? I think this is the problem as the beginning of my images are always fine as well, but then they get messed up during the middle of the image and the jpeg ends up smaller than it should be.

NVergunst, 2017-05-03:
What has been _mostly_ working for me, is a 2MHz SPI, reading roughly 125bytes at a time, with a variable (software state machine) delay between burst reads between 5ms and 10ms. 

I am guessing the way that the buffer code is written, when you request a pixel in a different page of memory, there is some non-zero delay required to allow the controller to catch up. In Arduino land where everything is dog slow, the user would never see these types of problems.

DallonTG, 2017-05-03:
Thanks. My code is using a 6 MHz SPI and the pictures I'm getting are in total around 200 kB. 

I might do some more testing and I'll post here if I learn anything useful.

NVergunst, 2017-05-20:
Ok, the logic inside the ArduCAM is DEFINITELY messed up. It is not my code. Verified by a super awesome calibrated scope. My software, my independent analyzer, my scope screenshots, and my scope's built in serial decode all agree. The only outlier is the ArduCAM. That's pretty telling.

I have it in a loop with a 2.2MHz SPI reading back the ID register which according to the datasheet will always 100% return 0x62. Well, it doesn't.

These 3 screencaptures show the reading of 0x64. Somehow it has skipped a clock and is outputting the wrong value, a clock too early. First image shows the previous transaction (good) plus the wrong one. Then zooming in on the bad 0x64 read. You can faintly see the cursors. I lined those up manually at the end of the first nibble to the bad data middle clock. Somewhere in that section the ArduCAM internally registered an additional clock. As you can see from the much more zoomed in capture, there is no spurious clock, and it looks the same as all the other reads.

[image]
[image]
[image]

And here is an even worse one! A readback of 0x00! It didn't even try...

[image]

There is something really wrong with the SPI interface. When transferring huge data like an image, small problems like this in pixels is hardly noticeable. But when it happens often... corruption.

NVergunst, 2017-05-20:
And it happens at 64KHz... Does not seem to be a speed problem. Read 0xC4 back which is 0x62 shifted a clock.

[image]

NVergunst, 2017-05-20:
I've been getting good stable reads at 256KHz SPI Clock with 10ms between reads. Crazy slow obviously, but right now I want to get it robust, then ratchet it in to find where it breaks. 

[image]
[image]

NVergunst, 2017-05-20:
At 1.5MHz with 40us between transactions, I get about 1 error per 20K transactions. Not good for a large image...

[image]
[image]

UCTRONICS, 2017-05-20:
@NVergunst 
Hi,
please try improve sampling frequency.

Let us know if you need more help.
Regards,
ArduCAM support team.

NVergunst, 2017-05-20:
Sampling us NOT the issue. This is a very deep memory scope and 2Gs/s. Believe me, that's not the issue.

supprot, 2017-05-22:
@NVergunst 
Hi,

That's a strange phenomenon.
We just do a test and read the OV5642 ID register  0x300a in the loop.
We set the SPI speed 4MHZ.
The data should be 0x56, if error , break.delay(3000) and it can work fine.
You can try the test code as below:

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif
// set GPIO16 as the slave select :
const int CS = 16;
ArduCAM myCAM(OV5642, CS);
void setup() {
uint8_t vid, pid;
uint8_t temp;
Wire.begin();
Serial.begin(115200);
Serial.println(F("ArduCAM Start!"));
// set the CS as an output:
pinMode(CS, OUTPUT);
// initialize SPI:
SPI.begin();
SPI.setFrequency(4000000); //4MHz
  
}
void loop() {
  while(1){
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    Serial.print("The vid is");
    Serial.println(vid,HEX);
    if(vid !=0x56) break;
    yield();
    }
    Serial.println("The SPI interface is error !!!");
    delay(3000);
}

Let us know if you need more help.
Reards,
ArduCAM support team.

NVergunst, 2017-05-22:
I don't have an Arduino hooked up to it, we use real hardware. Hardware that is proved to work in real embedded systems. You can see very plainly in the oscilloscope screenshots that a valid SPI request is going across the wire and it usually reads back 0x62. However sometimes it reads back 0x64 or 0xC4 or 0x00 or some other variation of bit shifting or just errors. 

You can see the waveforms. Captured by a calibrated expensive real lab hardware Agilent MSO scope.

NVergunst, 2017-05-22:
Well, back to getting garbage readings again. The data coming out of the SPI buffer is identical to my scope readings and what shows up in my micro's RAM. It's just not a valid image. I even installed a brand new from the box secondary unit. Same problems. 

[image]
[image]

ayoadejoe, 2017-05-23:
I don't know if this is similar, I set up two arduinos, one to work with the OV2640 mini, and the other to save to SD, and also carry out some other functions. I read the data from FIFO, then sent the byte data to the other arduino uno chip via Serial (SoftSerial). All was successful except that, the data received on the other arduino end which I saved to SDCard was though a valid JPEG file, was corrupted.
Also the nature of the byte data on both chip was different... the byte data was different no matter what I did.
So I just dumped the idea and found another way around to harvest the data.

pawcuq, 2017-10-17:
@ayoadejoe I'm experiencing same/similar problem, did you managed to figure it out?

pawcuq, 2017-10-17:
@NVergunst I'm experiencing same/similar problem, did you managed to figure it out?

pawcuq, 2017-10-17:
@NVergunst I'm experiencing same/silimar problem, did you manage to figure it out?

NVergunst, 2017-10-17:
> @NVergunst I'm experiencing same/silimar problem, did you manage to figure it out?

I abandoned the ArduCAM and switched to uCAM-III. After repeatedly being treated like an idiot by support I cannot recommend the ArduCAM at all. I make custom hardware all the time, and they don't believe it is possible there is anything wrong with their hardware ever. Magically when I switch to a different camera everything works exactly as it should. Meaning everything on my end is and was correct. Blaming it on my SPI peripheral because it is not in Arduino-land is ridiculous and obviously not the problem as I can use the SPI port to read and write megabytes to filesystems. 

The ArduCAM not only cost me a couple hundred bucks in hardware cost, but months in dev time. And it is 100% the ArduCAM problem. Drop it now, it will never work properly in any full system. It may be good enough for a noobie with a high school project, but it won't exist in real hardware.

pawcuq, 2017-10-17:
@NVergunst thanks for your quick response, I'll check out uCAM-III. STM32F103 here and I'm using hardware SPI (used the same configuration previously for many peripherals, LCDs etc without problem at all). I tried changing SPI frequency, number of bytes to read, delays between those reads, different JPEG sizes - corrupted JPEG will be transfered sooner or later no matter what. Ehh...

NVergunst, 2017-10-17:
That's why I captured the oscilloscope images. The SPI port (my peripheral like yours) is reading back exactly what is being put out on the bus. The problem is the data on the bus is wrong, that simple. The uCAM-III is a much lower resolution camera, but the interface has so far been solid. I can reliably get images out.

ArduCAM, 2017-10-18:
I'm sorry to hear that there is some problem using our cameras. I very regret to miss this thread from my email list. Our camera and examples are well tested on Arduino board, if you need to port to other platforms you have to contact us for help.

The following timing diagrams show the complete procedure how ArduCAM 5MP Plus (OV5642) module works:
1. Check the sensor model ID:
OV5642 I2C slave address  0x78 for write, 0x79 for read, OV5642 model ID register is 0x300A=0x56 and 0x300B=0x42
[image]
[image]
2. SPI operations based on Arduino UNO board at 4MHz SPI speed.
2.1 SPI Bus Test
Write 0x55 to 0x00 address and read the value from 0x00 address.
[image]
2.2 Image Data Transfer
After issue a capture command, pooling the capture completion flag by reading 0x41 register until euqals to 0x08.
[image]
Then read the image length from register 0x42~0x44, this is total length of the valid image data 
[image]
Next issue burst FIFO read command 0x3C and followed by bytes read from the FIFO, in the following diagram the 0xFF, 0xD8 are read from the beginning which is JPEG header in this case. Note each burst read cycle should be started with 0x3C command, followed by any numbers of data to be read. Here read 256 bytes in this example. When reading data, you have to send a dummy data over MOSI line, here sending 0x00 for example.
[image]
JPEG tail can also be read from the end of the valid data, 0xFF, 0xD9 in this case.
[image]
2.3 Complete capture and read timing diagram
The CS pin only active low during the 256 bytes burst read, CS pull high when send the 256 bytes over the UART. One frame has multiple burst read cycle.
[image]

3. Very important code in the burst read function
/************************important code in the burst read function**********************/
myCAM.CS_LOW();
myCAM.set_fifo_burst();
while ( length-- )
{
  temp_last = temp;
  temp =  SPI.transfer(0x00);
  //Read JPEG data from FIFO
  if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
  {
    buf[i++] = temp;  //save the last  0XD9     
    //Write the remain bytes in the buffer
    myCAM.CS_HIGH();
    Serial.write(buf,i);
    is_header = false;
    i = 0;
  }  
  if (is_header == true)
  { 
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else
    {
      //Write 256 bytes image data to file
      myCAM.CS_HIGH();
     Serial.write(buf,256);
     delay(5);
      i = 0;
      buf[i++] = temp;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
    }        
  }
  else if ((temp == 0xD8) & (temp_last == 0xFF))
  {
    is_header = true;
    Serial.println(F("ACK IMG"));
    buf[i++] = temp_last;
    buf[i++] = temp;   
  } 
}

4. Arducam host application 
[image]

5. Conclusion
We tried our best to implement read to use Arduino examples for user without a line of code.
If you have problem to modify our examples or port to other platforms, you need to email us for support. There may be some tricks during the development. Thanks for your understanding.

NVergunst, 2017-10-18:
Yes I get all that, and did all that in the code. As you can see, I was able to pull images. Pulling an image once was not a problem as some times it would work. The problem was in reliability. I would get bad data back from the module on the SPI line. The only suggestions were that my SPI peripheral was corrupt or bad, my entire architecture choice was bad, etc. And when I provided a screenshot as proof of the camera sending back bad data, the only suggestion was that my scope was not sampling at a high enough frequency or something else. Never is it the ArduCAM fault.

I am using professional high end equipment, and one of my previous jobs was a Validation Engineer to break digital interfaces attached to power regulation silicon that go into high end systems. IC level stuff. I am not just some dude trying out Arduino for the first time. I am using real hardware, in a real system, in real conditions, and the digital interface fails. 

Look at the May 20th post with the screenshot of the scope reading back 0xC4. That according to the datasheet is a read only register that ALWAYS returns 0x62. I should be able to read it at some maximum clock frequency back to back to back to back forever, and never get anything other than 0x62. Well there is plain view is a perfect 0xC4 read-back captured on analog waveform channels on a high end scope at 64KHz. I'd like an explanation of that first.

ArduCAM, 2017-10-19:
@NVergunst I understand that you are real engineer, I respect your feedback. I'm not saying there is no problem with ArduCAM hardware ourself. We used to have some hardware issue, like bit rotation issue on eariler version of 5MP cameras, and we have to use software work around to fix it. But now the new version hardware 5MP-Plus is much more stable. 
There is some tricks on programmig due to the complex burst reading bus timing, it will easily cause problem when break an entire image read into small bunches blob data burst read. You can also try single read command to see if it is stable. 
And we also find that the JPEG engine inside OV5642 sensor sometimes breakdown that cause broken JPEG data, maybe with the buffer overflow during compression. 
Also if you send the data over UART, you have to add some delay between each serial transfer.
Once again if you use other MCU other than Arduino, which we are not fully tested, you can contact us to work together. If you still have chance to try our new version, we can send you free sample for further testing.

NVergunst, 2017-10-19:
> @NVergunst I understand that you are real engineer, I respect your feedback. I'm not saying there is no problem with ArduCAM hardware ourself. We used to have some hardware issue, like bit rotation [...]

Unfortunately can't do a single read of a large image because I don't have the RAM, that's the reason I went with an ArduCAM and now a uCAM in the first place. I need something with the big RAM for images so my processor can fetch it slowly and pass it out over a cellular connection. When I got the image down to VGA quality I can store it in RAM but I was still getting tearing/blocking even with single read commands.

> And we also find that the JPEG engine inside OV5642 sensor sometimes breakdown that cause broken JPEG data, maybe with the buffer overflow during compression.

Well that would be good to know and maybe detect. Since you are wrapping the OV5642 digital firmware with your own, it would be good to know if a JPEG capture has the correct header and is at least the correct size or something like that. I obviously have no way of knowing if the data returned by the ArduCAM was corrupt because the data returned to it by the OV5642 was corrupt. However, this happened almost all the time, it tells me that's unlikely. Also the fact that reading an ArduCAM register that has no equivalent in the OV5642 was sometimes corrupted makes me believe the ArduCAM FPGA was causing corruption. **Did you see the scope shot of the bad data from a read only register that always returns 0x62 not returning 0x62 but bit-shifting it?** 

> Also if you send the data over UART, you have to add some delay between each serial transfer.

Everything was done over SPI. I also saw no mention of minimum time between packets in the datasheet for the SPI interface.

> Once again if you use other MCU other than Arduino, which we are not fully tested, you can contact us to work together. If you still have chance to try our new version, we can send you free sample for further testing.

There should be no reason for something with a universal interface like SPI to only work if you use a specific compiler. If so, then the compiler is doing is some magic in the background that should be in your datasheet because it is no longer generic SPI.

pawcuq, 2017-10-19:
@ArduCAM to sum up - you have virtually no datasheet. Scopes which you provided should be in much more detail provided in DS from day one. Like said in Halt and Catch Fire - "Software goes, hardware stays".

pawcuq, 2017-10-19:
@ArduCAM to sum up - you have virtually no datasheet. Scopes which you provided should be in much more detail provided in DS from day one. Like said in Halt and Catch Fire - "Software comes and goes. Hardware is forever.".

supprot, 2017-10-20:
@pawcuq

This is the datasheet for Arducam www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Plus_OV5642_Camera_Module_DS.pdf
http://www.arducam.com/downloads/shields/ArduCAM_Software_Application_Note.pdf
www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Plus_Camera_Shield_Hardware_Application_Note.pdf

ArduCAM, 2017-10-24:
@NVergunst 
I guess you got the old 5MP camera, can you send me the photo to confirm.
We can send you a free new camera for your double checking, or you can send us your testing code, we can help you test here.
Thanks.

ghgv, 2018-11-06:
Hi. 
I got a 5MP Mini OV5642 module that I purchased recently. I see the software in an Arduino UNO works with the supplied Host App. But when I try to make my own app  I get the same behaviour as described above. I wonder what the problem could be. And I wonder if I can take a look at the source code of the Host App. I see, it really changes a lot of settings in the serial port before it starts transmitting.