ArduCAM/Arduino #169: OV5642 blurr images with movement

wdouglas1999, 2017-04-10:
Hi, I am too evaluating the OV5642 and have the same blurry images with movement. I have the version with 8Mbyte FIFO. My test program is as follows.
I capture multiple images until the fifo is full.
Then I copy the images to SD card.
When I look at the images I see a few blurry images when there is movement.
From my observation since the images are stored directly to the FIFO it seems that this is an issue with the arducamchip and OV5642.
Has anyone managed to get this fixed some how?

supprot, 2017-04-11:
@wdouglas1999 
Hi,
Would you please attach us some pictures for the  phenomenon?
Let us know if you need more help.

Best regards,
ArduCAM support team.

wdouglas1999, 2017-04-11:
Hi thank you for the reply. Attached are my pictures. Images come straight from the FIFO. Resolution is 640x480. When there is movement it is blur. Any way you can help me to solve this issue?

[image]
[image]

supprot, 2017-04-11:
@wdouglas1999 
Hi,
Have you used our ArduCAM_Mini_5MP_Plus_short_movie_clip demo?
Do you changed the #define   rate     0x0a ?

Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-11:
Hi, no I have not used this demo. Will the demo work at 640 x 480?

supprot, 2017-04-11:
@wdouglas1999 
Hi,
Yes,but you should change the resolution in the AVI header.
Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-11:
Do I change the resolution with this line?
  myCAM.OV5642_set_JPEG_size(OV5642_320x240) ?
To   myCAM.OV5642_set_JPEG_size(OV5642_640x480)

Is this all that needs to be changed?
Thx

supprot, 2017-04-11:
@wdouglas1999 
Hi,
You should not only change  myCAM.OV5642_set_JPEG_size(OV5642_320x240) ?
To myCAM.OV5642_set_JPEG_size(OV5642_640x480) but also change the AVI header in the 
const int avi_header[AVIOFFSET] {....}.
Please refer to this configure:
const int avi_header[AVIOFFSET] PROGMEM ={
  0x52, 0x49, 0x46, 0x46, 0xD8, 0x01, 0x0E, 0x00, 0x41, 0x56, 0x49, 0x20, 0x4C, 0x49, 0x53, 0x54,
  0xD0, 0x00, 0x00, 0x00, 0x68, 0x64, 0x72, 0x6C, 0x61, 0x76, 0x69, 0x68, 0x38, 0x00, 0x00, 0x00,
  0xA0, 0x86, 0x01, 0x00, 0x80, 0x66, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x84, 0x00, 0x00, 0x00,
  0x73, 0x74, 0x72, 0x6C, 0x73, 0x74, 0x72, 0x68, 0x30, 0x00, 0x00, 0x00, 0x76, 0x69, 0x64, 0x73,
  0x4D, 0x4A, 0x50, 0x47, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x00, 0x00, rate, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x73, 0x74, 0x72, 0x66,
  0x28, 0x00, 0x00, 0x00, 0x28, 0x00, 0x00, 0x00, 0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0xf0, 0x00,0x00, 0x00,
  0x01, 0x00, 0x18, 0x00, 0x4D, 0x4A, 0x50, 0x47, 0x00, 0x84, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54,
  0x10, 0x00, 0x00, 0x00, 0x6F, 0x64, 0x6D, 0x6C, 0x64, 0x6D, 0x6C, 0x68, 0x04, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x00, 0x01, 0x0E, 0x00, 0x6D, 0x6F, 0x76, 0x69,
};
Let us know if you need more help.
Regards,
ArduCAM support team.

supprot, 2017-04-11:
@wdouglas1999 
Hi，
sorry for the first configure,
Please refer to this below;
const int avi_header[AVIOFFSET] PROGMEM ={
  0x52, 0x49, 0x46, 0x46, 0xD8, 0x01, 0x0E, 0x00, 0x41, 0x56, 0x49, 0x20, 0x4C, 0x49, 0x53, 0x54,
  0xD0, 0x00, 0x00, 0x00, 0x68, 0x64, 0x72, 0x6C, 0x61, 0x76, 0x69, 0x68, 0x38, 0x00, 0x00, 0x00,
  0xA0, 0x86, 0x01, 0x00, 0x80, 0x66, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x84, 0x00, 0x00, 0x00,
  0x73, 0x74, 0x72, 0x6C, 0x73, 0x74, 0x72, 0x68, 0x30, 0x00, 0x00, 0x00, 0x76, 0x69, 0x64, 0x73,
  0x4D, 0x4A, 0x50, 0x47, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x00, 0x00, rate, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x73, 0x74, 0x72, 0x66,
  0x28, 0x00, 0x00, 0x00, 0x28, 0x00, 0x00, 0x00, 0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0x00, 0x00,
  0x01, 0x00, 0x18, 0x00, 0x4D, 0x4A, 0x50, 0x47, 0x00, 0x84, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54,
  0x10, 0x00, 0x00, 0x00, 0x6F, 0x64, 0x6D, 0x6C, 0x64, 0x6D, 0x6C, 0x68, 0x04, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x00, 0x01, 0x0E, 0x00, 0x6D, 0x6F, 0x76, 0x69,
};

Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-11:
Hi! Thanks again for all your help. The example works fine at 640 x 480, however there is still a ghosting effect and the images are not clear when there is movement. As the images are loaded directly from the sensor to the FIFO by the Aducam chip I am not sure how I will be able to resolve this issue. Is this the normal performance of the camera ? Can anything else be done?

supprot, 2017-04-12:
@wdouglas1999 
Hi, 
It is related to the frame rate of cameras,What about 320x240 resolution?
The big resolution will make our 5MP 's frame low.
Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-12:
Hi, the 320 x 240 is better, but no 100%. This module is expensive for only this capability. Why is frame rate low if the module has the RAM to buffer all the images from the sensor? Then the aducamchip is too slow? 
It is strange because the frame rate looks fast enough but the image has ghosting. I've worked with the VC 0706 chip and when the frame rate is low the movement is jerky but now ghosting effect. I was hoping your camera would perform better since it has all that RAM onboard. Do you think it could be the image sensors exposure settings or do you think it is purely a frame rate issue? Anyway, the product is still great but just not quite there yet. Thx

supprot, 2017-04-12:
@wdouglas1999 
Hi,
We just give you a demo and we don't configure the register in detail.
It is related to the register configure.
You should configure relevant register to increase the frame rate which 
mat take you some time.
Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-12:
Hi, ok thank you. I will play around and see if I can make progress.

supprot, 2017-04-12:
@wdouglas1999 
Hi,
The performance of the processor is also a factor to influence the frame rate.
Regards,
ArduCAM support team.

wdouglas1999, 2017-04-12:
Hi, thank you for the feedback. After reading the application note on the OV5642 it seems that they do mention that the rolling shutter of the sensor typically causes this type of issues (smearing effect) if the frame rate is not high enough. So it may be that nothing can be done to resolve this issue from a user point of view.

wdouglas1999, 2017-04-12:
Hi, one more question. Can arducam chip work with "frame exposure mode" because frame exposure mode is different to rolling exposure mode.
Do you have a command for this maybe?
Thank you again

supprot, 2017-04-12:
@wdouglas1999 
Hi,
We are optimizing and developing more functions for the OV5642 camera.
Let us know if you need more help.
Regards,
ArduCAM support team.