ArduCAM/Arduino #170: OV2640 dark picture after sensor standby

Schnulli87, 2017-04-11:
Hi,
I have a question regarding the usage of low power mode (GPIO_PWDN_MASK):
I have modified the standard web-sketch to put the camera into sensor standby after sending a photo and wake it up, when a http-request for a picture is made.
Here is what I have discovered:
If there is no delay between leaving standby and starting a capture, I receive a completely white image.
If I add a delay of 100ms, the image is almost black with some areas recognizable where there should be bright objects in the photo.
The longer the delay after standby is, the better the image gets, but the colors are still shifted towards green and yellow. I have tried with delays up to 2 seconds, but the image still did not have the correct colors/white balance.
What can/must I do to improve this behavior? Fixed white balance? If yes: how?
How long does the sensor need after leaving standby, until the white balance is correct?
In the sensor datasheet, I could not find any information regarding delays after standby.
If i keep the sensor awake, every image has the correct colors, independent of the type of light source used.

Thank you in advance!

supprot, 2017-04-12:
@Schnulli87 
Hi,

Yes,you should delay some time for the sensor back to normal From low power.When you  set GPIO_PWDN_MASK,the camera go to power down mode and the power of the sensor is very low.
When you reset the GPIO_PWDN_MASK bit, you shoud give some time to wait the camera back normal.
The specific delay time should be adjusted according to actual situation.

Let us know if you need more help.
Regards,
ArduCAM support team.

Schnulli87, 2017-04-12:
Thanks for your quick answer.
Would it be possible to read the current values regarding white balance, exposure, ... from the sensor, store them in the Arduino/ESP, then put the sensor into standby mode and when the camera is needed again, wake the sensor from standby and write the old white balance, exposure, ... values back to the sensor? 

If this is possible, the sensor could start re-adjusting from these values instead of from the "standby values" and the delay could be much shorter. Maybe I'm thinking to simple here...

Which of the functions would I have to call and with which parameters?

The reason why I do not want the delay (it would have to be longer than 2 seconds as I noticed) is, that the capture is initiated via the http request and I do not know how the device requesting the image would handle such a long delay. As I request an image every two minutes, I guess that the values for white balance exposure, .. would not change very much from one photo to the next.

supprot, 2017-04-13:
@Schnulli87 
Hi,
Maybe this function can help you.
void ArduCAM::set_Brightness(uint8_t Brightness)
{
	switch(Brightness)
	{
		case Brightbess4:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x40);
			wrSensorReg8_8(0x7d, 0x00);
		#endif	
		break;
		case Brightbess3:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x30);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;	
		case Brightbess2:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x20);
			wrSensorReg8_8(0x7d, 0x00);
		#endif	
		break;
		case Brightbess1:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x10);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;
		case Brightbess0:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x00);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;	
	}
		
}Let us know if you need more help.
Regards,
ArduCAM support team.

jahearnco, 2017-07-19:
Hello. I'm also having darkness issues, and tried to use your brightness options above, but I doubt the issue is brightness related ...

I've been using and testing the OV2640 for over 2 months now, and the greatest changes I've seen come about with the lighting options, not the brightness options. I don't think I have the correct lighting options to choose from. All of them are too dark for indoors - can you provide lighting options in the same way you provided brightness options?

One thing of interest to note: recently I made a mistake with my power supply - a LiPo battery cell that was supplying the same voltage to my esp8266 AND my OV2640. I let the voltage drop below 2.8 volts  and was still able to view pictures from the OV2640 uploaded to my internet server.  The pictures were no longer dark. THEY WERE PERFECT!! The lighting and brightness and clarity were perfect. Any idea why the lower voltage fixed the picture quality??

Thanks!
John

Schnulli87, 2017-07-23:
Hi John,
I'm sorry, I didn't post my results  of supprots proposals.
I also had no luck with the brightness options. What DID work for me, was to change the white balance values. This is giving me good pictures from morning to evening since serveral months now.
My code is below.
I have two functions for getting a picture, as sometimes the image is damaged after transfer and I need to check for this on the PC. I have one function (serverNewCapture() ) for taking a picture and sending it. The other function (serverReplayCapture() ) sends the last image from the buffer again, or takes a new one if no picture has been taken before (after power up/reset).
serverNewCapture() wakes the sensor up, writes the white-balance and similar stuff to the sensor, then takes a picture, sends it and puts the sensor back to sleep.
serverReplayCapture() does not wake or put the sensor to sleep. It just resets the FIFO position and then sends the stored image.
**Important!
If I remember it correctely, there is a typo somewhere in the library or the chip documentation. The masks for sensor standby and fpga-standby are switched somewhere. I think this is the reason, why I write 0x02 instead of the GPIO_PWDN_MASK in serverNewCapture().
**
Looking through the code, I just noticed, that I still have several delays and debug output included. It seems, that the camera is running all the time with this. You can try to remove the 1000ms delay as it was used to see if the chip changed the white balance on its own, if I wait some time. Also the get_AWB() is  only for testing purposes.

`
Global:
void serverNewCapture(void);
void serverReplayCapture(void);
void restore_Settings(void);
void get_AWB(void);

unsigned char val_c7;
unsigned char val_cc;
unsigned char val_cd;
unsigned char val_ce;

// #################################
void start_capture()
{
  myCAM.clear_fifo_flag();
  myCAM.start_capture();
}

void serverNewCapture()
{
  myCAM.clear_bit(ARDUCHIP_GPIO,0x02);                    // wake Sensor
  delay(100);
  restore_Settings();
//  myCAM.set_format(JPEG);
  get_AWB();
  Serial.println("0xc7 0xcc 0xcd 0xce before delay");
  Serial.println(val_c7,HEX);
  Serial.println(val_cc,HEX);
  Serial.println(val_cd,HEX);
  Serial.println(val_ce,HEX);
  delay(1000);
  get_AWB();
  Serial.println("0xc7 0xcc 0xcd 0xce after delay");
  Serial.println(val_c7,HEX);
  Serial.println(val_cc,HEX);
  Serial.println(val_cd,HEX);
  Serial.println(val_ce,HEX);

  myCAM.clear_fifo_flag();                                            // clear "picture available"-Flag
  myCAM.start_capture();                                              // capture new picture
  Serial.println("Capturing");

  while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)) yield();
  
  Serial.println("Capture Done!");
  camCapture(myCAM);
  Serial.println("Capture sending done, going to sleep!");
    get_AWB();
  Serial.println("0xc7 0xcc 0xcd 0xce after capture");
  Serial.println(val_c7,HEX);
  Serial.println(val_cc,HEX);
  Serial.println(val_cd,HEX);
  Serial.println(val_ce,HEX);

  myCAM.set_bit(ARDUCHIP_GPIO,0x02);                      // put sensor into Standby
}

[... 77 lines cut ...]
  myCAM.wrSensorReg8_8(0x7d, 0x02);
  myCAM.wrSensorReg8_8(0x7c, 0x03);
  myCAM.wrSensorReg8_8(0x7d, 0x48);
  myCAM.wrSensorReg8_8(0x7d, 0x48);

  // Contrast 0
  myCAM.wrSensorReg8_8(0xff, 0x00);
  myCAM.wrSensorReg8_8(0x7c, 0x00);
  myCAM.wrSensorReg8_8(0x7d, 0x04);
  myCAM.wrSensorReg8_8(0x7c, 0x07);
  myCAM.wrSensorReg8_8(0x7d, 0x20);
  myCAM.wrSensorReg8_8(0x7d, 0x20);
  myCAM.wrSensorReg8_8(0x7d, 0x20);
  myCAM.wrSensorReg8_8(0x7d, 0x06);

  // Image type: Normal
  myCAM.wrSensorReg8_8(0xff, 0x00);
  myCAM.wrSensorReg8_8(0x7c, 0x00);
  myCAM.wrSensorReg8_8(0x7d, 0x00);
  myCAM.wrSensorReg8_8(0x7c, 0x05);
  myCAM.wrSensorReg8_8(0x7d, 0x80);
  myCAM.wrSensorReg8_8(0x7d, 0x80);
}

`

supprot, 2017-07-24:
@Schnulli87 
Hi,

The camera is automatic exposure default, so when you change the resolutions or make it normal from standby mode.The sensor need some time to adapt to it. So when you change it's resolutions or make it normal from standby mode,it is better to add 1 to 3 seconds delay.

Please refer to this link to know more problems' asnwers.
http://www.arducam.com/knowledge-base/arducam-typical-problems-arduino-uno/?preview_id=1720&preview_nonce=f1e4612837&preview=true

Let us know if you need more help.
Regards.