ArduCAM/Arduino #171: Arducam OV2640 motion detector

rmcncal, 2017-04-12:
Hi

I'm trying to get a OV2640 on an Arducam ESP8266 working with a PIR motion detector to take a picture when it detects movement. I have the motion detector trigger wire connected to GPIO12 and running the code below. 

I use

 attachInterrupt(12, intruder_detect, CHANGE);  in setup()

on pin 12 to trigger an intruder detection event which then captures a photo.

I get in my serial monitor (after pressing reset a couple of times). No pictures are ever captured
Code is a modified version of ArduCAM_ESP8266_V2_OV2640_Capture2SD along with interrupt for motion detector. Any ideas why it's not working?

2dN,4`lMMØlb8óÿArduCAM Start!
85
SD Card detected!
OV2640 detected.
2dO,4`lMMØlb8óÿArduCAM Start!
Intruder detected
star Capture
¶Eô|vAM¬tè4C(ArduCAM Start!
85
SD Card detected!
OV2640 detected.
$lÍ+`,lløp$:òûArduCAM Start!
Intruder detected
star Capture
¿„FJ
ô@HàþArduCAM Start!
85
SD Card Error
OV2640 detected.

#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;
//Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
ArduCAM myCAM(OV2640, CS);

int speakerOut = 12;//Piezo buzzer's positive terminal is connected to digital pin 9               
byte names[] = {'c', 'd', 'e', 'f', 'g', 'a', 'b', 'C'};  
int tones[] = {1915, 1700, 1519, 1432, 1275, 1136, 1014, 956};
byte melody[] = "2d2a1f2c2d2a2d2c2f2d2a2c2d2a1f2c2d2a2a2g2p8p8p8p";
// count length: 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0
//                                10                  20                  30
int count = 0;
int count2 = 0;
int count3 = 0;
int MAX_COUNT = 24;
int statePin = LOW;

const int motionSensorPin = 12;

void siren();

volatile byte intruder;

void myCAMSaveToSDFile(){
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  static int n = 0;
  uint8_t temp, temp_last;
  File file;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println("star Capture");
 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println("Capture Done!");  

 //Construct a file name
 k = k + 1;
 itoa(k, str, 10);
 strcat(str, ".jpg");
 //Open the new file
 file = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 if(! file){
  Serial.println("open file faild");
  return;
 }
 i = 0;
 myCAM.CS_LOW();
 myCAM.set_fifo_burst();
#if !(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM))
SPI.transfer(0xFF);
#endif
 //Read JPEG data from FIFO
 while ( (temp !=0xD9) | (temp_last !=0xFF)){
  temp_last = temp;
  temp = SPI.transfer(0x00);

  //Write image data to buffer if not full
  if( i < 256)
   buf[i++] = temp;
   else{
    //Write 256 bytes image data to file
    myCAM.CS_HIGH();
    file.write(buf ,256);
    i = 0;
    buf[i++] = temp;
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();
   }
   delay(0);  
 }
 
 //Write the remain bytes in the buffer
 if(i > 0){
  myCAM.CS_HIGH();
  file.write(buf,i);
 }
 //Close the file
 file.close();
  Serial.println("CAM Save Done!");
}

void setup()
 {

  attachInterrupt(12, intruder_detect, CHANGE);//Initialize the intterrupt pin for the motion sensor (Arduino digital pin 2)
 

   uint8_t vid, pid;
   uint8_t temp;
   Wire.begin();
   
   Serial.begin(9600);
   Serial.println("ArduCAM Start!");
   //Serial.begin(115200);
  
 
   //set the CS as an output:
  pinMode(CS,OUTPUT);

  //initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHZ

  delay(1000);
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  Serial.println(temp);
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");
    while(1);
  }
  
    //Initialize SD Card
  if(!SD.begin(SD_CS)){
    Serial.println("SD Card Error");
  }
  else
  Serial.println("SD Card detected!");
  

  
//Check if the camera module type is OV2640
  myCAM.wrSensorReg8_8(0xff, 0x01);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  if ((vid != 0x26 ) && (( !pid != 0x41 ) || ( pid != 0x42 )))
   Serial.println("Can't find OV2640 module!");
   else
   Serial.println("OV2640 detected.");
   myCAM.set_format(JPEG);
   myCAM.InitCAM(); 
   
   
   
 }

 void loop()
 {
  
 
 }
 
 void intruder_detect()//This function is called whenever an intruder is detected by the arduino
 {
   intruder++;
   Serial.println("Intruder detected");

   myCAMSaveToSDFile(); 

    
 }

supprot, 2017-04-12:
@rmcncal 
Hi,
Your SD card is error,
The GPIO12 is the MISO signal.

Best regards,
ArduCAM support team.

rmcncal, 2017-04-13:
Hi

The SD card is fine as it  captures photos when I run ArduCAM_ESP8266_V2_OV2640_Capture2SD
it's when I modify this sketch to use a PIR motion detector that it doesn't work, I have tried GPI02 & GPI016 pins without success, do you know which pin  I can use to process an output signal from a PIR motion detector? I use an interrupt to capture the output signal to take a photo.

supprot, 2017-04-13:
@rmcncal 
Hi,

If you use both SD card and camera module on our ArduCAM ESP8266 UNO board, 
you will don't have other GPIO pin for PIR input.

Let us know if you need more help.
Regards,
ArduCAM support team.

watkinsllc, 2017-05-14:
Hello,  
I am trying the same thing and have found 2 issues. One is the wifi ESP8266 which is not recognized when you type AT in the serial monitor at any baud rate. And 2. the PIR cannot be used if you want to use the SD card with this board.
So, do you recommend using the Arduino UNO R3 with a ESP8266-01 with the PIR and OV2640. If so, what is the pinout for this and what programs should be used?
Thanks