ArduCAM/Arduino #172: OV5642 Corrupt JPEG

Mschindl301, 2017-04-12:
Hello,
I am using an Arduino Mega 2560 with an OV5642 Mini Rev B. I have uploaded the Mini_Video_Streaming example to the arduino and when I open the host app, the camera is detected and SPI is ok. I select single photo and 320x240 picture, but when I click capture I get about 7800 bytes of data, but only saves a file 1/5th of the time. When it does save the file, it is a corrupt jpeg.

I did have to change one thing in the code. I changed Serial.begin(921600) to Serial.begin(115200). If I don't I don't get anything displayed in the host app besides COM open. I am using an Arduino Mega with a CH340g as the USB to Serial converter because it is a cheap clone. Does that have anything to do with the need to change the baud rate, or am I missing something else as well?

Thank you for the help!!

supprot, 2017-04-12:
@Mschindl301 
Hi,
Please attached us some pictures about your phenomenon.

Regards,
ArduCAM support team.

Mschindl301, 2017-04-12:
I have attached two corrupted pictures below, thank you.
[image]
[image]

dmiddlecamp, 2017-04-18:
I'm also getting corrupted photos, but I'm trying to port this to run on a Particle Photon.  I would expect the start of the JPEG images to be somewhat consistent, but they're each quite different, is it possible the JPEG header is missing?

[image]
[image]
[image]

supprot, 2017-04-19:
@Mschindl301 @dmiddlecamp
Hi, there isn't jpeg header in the picture you attached us.
The head of JPEG begin from 0xFF 0xD8 and the end with 0xFF 0xD9.
If the datas doesn't have the jpeg header, maybe your camera init has problem.
Does your camera work fine on other platform?

Let us know if you need more help.
Regards,
ArduCAM support team.

dmiddlecamp, 2017-04-19:
I'll test both my units on an arduino as well, but if I'm getting the right VID/PID back from the device, then I'm assuming I'm communicating with it correctly?  Am I missing delays or other commands in here?

    // general init

    myCAM.set_format(JPEG);
    myCAM.InitCAM();
    myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
    myCAM.clear_fifo_flag();
    myCAM.write_reg(ARDUCHIP_FRAMES,0x00);
    myCAM.OV5642_set_JPEG_size(OV5642_320x240);

    // before capture

    myCAM.flush_fifo();
    myCAM.clear_fifo_flag();
    myCAM.start_capture();

    // waiting for photo

    while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

    // reading photo contents

    int length = myCAM.read_fifo_length();
    while (bytesRead < length)
    {
          temp = myCAM.read_fifo();
          delayMicroseconds(12);
    }

    ...
    //clear fifo queue
    myCAM.clear_fifo_flag();

Is this roughly on track?  Am I missing anything?  I tried to piece this together from examples

dmiddlecamp, 2017-04-19:
Turns out porting the platform changes to the latest library with some hacks and extra timing worked well.  I've got valid photos coming through which is very exciting :)  but now I'm seeing the streaking / striping at higher resolutions

supprot, 2017-04-19:
@dmiddlecamp 
Hi,
Congratulation to you to solve the problem before.
When the resolution is higher, it will transmit more image datas by serial port.
So you should ensure the hardware connection is stable.We suggest you use our
adapter board.

Let us know if you need more help.
Regards,
ArduCAM support team.

dmiddlecamp, 2017-04-19:
Hi @supprot,

  After further review and debugging, I've discovered the choppy images to be a result of a poor physical SPI connection with the ArduCAM module.  Making a more reliable physical connection allows for extracting full quality images.

Thanks!
David

Mschindl301, 2017-04-19:
@dmiddlecamp 
I still seem to be getting corrupt images through my ArduCam. You said you just downloaded the most recent library and changed some delays? I have tried the same and still no luck, can you tell me exactly what you changed?
Thanks!!

dmiddlecamp, 2017-04-19:
Hi @Mschindl301,

  I found the older library was giving me corrupted data, but the newer library with the initialization and delay schemes, with a set SPI transaction speed of 8Mhz is working well.

Here's the firmware I'm running, the delays are important I think...?

https://github.com/dmiddlecamp/particle_arducam/blob/master/firmware/tcp_photographer/main.ino#L107-L132

SPI Transactions were necessary to get the SPI stuff working correctly with the newer library for Particle:

https://github.com/dmiddlecamp/particle_arducam/commit/bdfc39f177ccfd0b8d93374549b96784cd686aeb#diff-421c2cc55e7cb6f15d8b5115a808af8f

With the delays and the newer library, and good physical electrical connections, I started getting good data out.