ArduCAM/Arduino #173: Mini-5MP-Plus FPS limit

michprev, 2017-04-12:
Hey,
I am running ArduCAM_Mini_5MP_Plus_Video_Streaming sketch and I cannot get over 6 FPS on Arduino UNO. I have also rewritten the code for STM32 (source available [here](https://github.com/michprev/Drone/tree/master/Camera/src)). The issue is that waiting for CAP_DONE_MASK of ARDUCHIP_TRIG takes too long (in average about 300 ms).

The solution would be to use the short video capture feature but this does not allow to stream the video in real time.

Is there any way how to get higher FPS?
Thank you!

supprot, 2017-04-13:
@michprev 
Hi,
Please try to teat this demo:
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It can  continue shooting  and store it into the SD card  in AVI format
// The demo sketch will do the following tasks
// 1.Shoot video button, began to shoot video 
// 2. Set the camera to JPEG output mode.
// 3. Capture a JPEG photo and buffer the image to FIFO
// 4.Write AVI Header
// 5.Write the video data to the SD card
// 6.More updates AVI file header
// 7.close the file
// The file header introduction
// 00-03 :RIFF
// 04-07 :The size of the data
// 08-0B :File identifier
// 0C-0F :The first list of identification number
// 10-13 :The size of the first list
// 14-17 :The hdr1 of identification
// 18-1B :Hdr1 contains avih piece of identification 
// 1C-1F :The size of the avih
// 20-23 :Maintain time per frame picture
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV5640_MINI_5MP_PLUS or OV5642_MINI_5MP_PLUS platform.
#if !(defined (OV5640_MINI_5MP_PLUS)||defined (OV5642_MINI_5MP_PLUS))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

#define   FRAMES_NUM    0x07
#define   rate     0x0a
#define SD_CS 9
#define KEY 2
#define AVIOFFSET 240
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
uint32_t total_time = 0;
unsigned long movi_size = 0;
unsigned long jpeg_size = 0;
const char zero_buf[4] = {0x00, 0x00, 0x00, 0x00};
const int avi_header[AVIOFFSET] PROGMEM ={
  0x52, 0x49, 0x46, 0x46, 0xD8, 0x01, 0x0E, 0x00, 0x41, 0x56, 0x49, 0x20, 0x4C, 0x49, 0x53, 0x54,
  0xD0, 0x00, 0x00, 0x00, 0x68, 0x64, 0x72, 0x6C, 0x61, 0x76, 0x69, 0x68, 0x38, 0x00, 0x00, 0x00,
  0xA0, 0x86, 0x01, 0x00, 0x80, 0x66, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x84, 0x00, 0x00, 0x00,
  0x73, 0x74, 0x72, 0x6C, 0x73, 0x74, 0x72, 0x68, 0x30, 0x00, 0x00, 0x00, 0x76, 0x69, 0x64, 0x73,
  0x4D, 0x4A, 0x50, 0x47, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x00, 0x00, rate, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x73, 0x74, 0x72, 0x66,
  0x28, 0x00, 0x00, 0x00, 0x28, 0x00, 0x00, 0x00, 0x80, 0x02, 0x00, 0x00, 0xe0, 0x01, 0x00, 0x00,
  0x01, 0x00, 0x18, 0x00, 0x4D, 0x4A, 0x50, 0x47, 0x00, 0x84, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54,
  0x10, 0x00, 0x00, 0x00, 0x6F, 0x64, 0x6D, 0x6C, 0x64, 0x6D, 0x6C, 0x68, 0x04, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x00, 0x01, 0x0E, 0x00, 0x6D, 0x6F, 0x76, 0x69,
};
void print_quartet(unsigned long i,File fd){
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
[... 238 lines cut ...]
print_quartet(movi_size +12*frame_cnt+4, outFile);//riff file size
//overwrite hdrl
unsigned long us_per_frame = 1000000 / rate; //(per_usec); //hdrl.avih.us_per_frame
outFile.seek(0x20);
print_quartet(us_per_frame, outFile);
unsigned long max_bytes_per_sec = movi_size * rate/ frame_cnt; //hdrl.avih.max_bytes_per_sec
outFile.seek(0x24);
print_quartet(max_bytes_per_sec, outFile);
unsigned long tot_frames = frame_cnt;    //hdrl.avih.tot_frames
outFile.seek(0x30);
print_quartet(tot_frames, outFile);
unsigned long frames =frame_cnt;// (TOTALFRAMES); //hdrl.strl.list_odml.frames
outFile.seek(0xe0);
print_quartet(frames, outFile);
outFile.seek(0xe8);
print_quartet(movi_size, outFile);// size again
//Close the file
outFile.close();
is_header = false;
Serial.println(F("Movie save OK"));
return 1;
}
Let us know if you need more help.
Regards,
ArduCAM support team.

michprev, 2017-04-13:
This does not allow to stream the video. We have to wait for about 5 minutes before it is possible to load images from the buffer. I am looking for a solution that would allow to stream the video (images) with no delay and higher FPS.

supprot, 2017-04-14:
@michprev 
Hi,
The speed of frame  is related to register configuration. In this demo ,after you  button down the key,it will start capture until 8M datasand then save it to SD card,We just give you a demo and the frame speed can uo to ten fps.You can adjust the relevant register to change the frame speed. 

Let us know if you need more help.
Regards,
ArduCAM support team.

michprev, 2017-04-17:
@supprot
Hi,
I think that I understand now - I need to tune camera registers over I2C. Are there available any schematics for Arducam Mini Plus boards (I am interested in OV5640 and OV5642 boards)? I don't know how works communication between the camera and the FIFO and I don't want to break it by setting improper value into a camera register.
Is there any description of OV5640 and OV5642 registers or do I need to contact Omnivision support?

Thank you!

supprot, 2017-04-17:
@michprev 
Hi,

Please refer to the below link:
www.uctronics.com/download/OV5642_camera_module_software_application_notes_1.1.pdf
www.uctronics.com/download/OV5642_CSP_DS_1.11_OVT（5M EDoF AFC）.pdf

Let us know if you need more help.
Regards,
ArduCAM support team.

supprot, 2017-04-17:
@michprev 
Hi,
http://www.uctronics.com/download/OV5642_CSP_DS_1.11_OVT%EF%BC%885M%20EDoF%20AFC%EF%BC%89.pdf
Let us know if you need more help.
Regards,
ArduCAM support team.

michprev, 2017-04-17:
@supprot

Thank you very much!!
Hope to get it working soon.

supprot, 2017-04-17:
@michprev 
Hi,

You are welcome, please feel free to tell us if you have any problems.

Regards,
ArduCAM support team.

ShilRenz, 2018-02-16:
@supprot 
I have the same issue, I have read www.uctronics.com/download/OV5642_camera_module_software_application_notes_1.1.pdf, but I don't know where to put this codes 
[image], may I know where will I put this codes?
Thank you!

supprot, 2018-02-19:
@ShilRenz
Hi,
You should put this configure to void ArduCAM::InitCAM() which is in ArduCAM.cpp file.

ShilRenz, 2018-02-19:
@supprot 
thank you for so much for helping, hope this might help me with my project.

ShilRenz, 2018-02-24:
I tried to put the code inside void ArduCAM::InitCAM() but during the compilation there was an error I encountered.
[image]
i put the code inside case ov5642
[image]

supprot, 2018-02-24:
@ShilRenz
Hi,
Please replace 'write_i2c()' with 'wrSensorReg16_8()'.

ShilRenz, 2018-02-24:
thank you 
@supprot 
it quite faster now than before.

supprot, 2018-02-24:
@ShilRenz 
Hi,
How mant fps can you get?

supprot, 2018-02-24:
@ShilRenz 
How many fps can you get?

ShilRenz, 2018-03-01:
@supprot 
i got 2-3 fps
can i make it more faster up to 15 fps?

supprot, 2018-03-02:
@ShilRenz 
This has a lot to do with the speed of the controller and the speed of the SPI.
You can try it on STM32 controller.

ShilRenz, 2018-03-02:
@supprot 
i don't have STM32 controller, is there any other way to change the fps in my arducam? it was directly connected to my arduino.
do you have facebook account, may i contact you there? so that I  can ask you easily when you are on line.

supprot, 2018-03-02:
@ShilRenz 
Hi,
Facebook is not available for the time being.Do you have Skype?
If so, you can contact us and our Skype ID is fpga4u

ShilRenz, 2018-03-04:
@supprot
ok sir, thank you for the reply, I don't  have Skype.  I'll just  here or to the other issues if I have concerns. thank you a kot for your help. .😊

dcavone, 2022-05-18:
Hello,
the time between start_capture() and the flag CAP_DONE_MASK = 1 depends on only OV5642 sensor?
Thank you

Bjmcontrols2021, 2025-04-24:
Hello, 

I've read through this thread looking to increase the capture rate of my OV5642.  I am only requiring an image every 10 seconds.  So I am not concerned with the time it takes to transfer the image to my ESP32 but I want to capture the image into the camera's buffer as quickly as possible, i.e., 30fps or faster if possible.  Unfortunately I've only been able to acheive 6fps.  Is it possible to do better?

Thnaks