ArduCAM/Arduino #174: SD Card light not turning on

irowebbn, 2017-04-14:
Hello, I am using a ArduCAM 4 Camera Adapter Board, testing with one OV5642 Camera and a [Sparkfun RedBoard](https://www.sparkfun.com/products/13975) (with the same compatibility as the Arduino Uno R3)

I am trying to run the example ArduCAM_Shield_V2_Capture2SD.ino 
 ( ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Capture2SD ), but it will not write any photos to the SD card, and furthermore, the LED indicator beside the SD card slot (labeled SD, and presumably to indicate that the SD card slot is in use) never comes on whether or not there is a microSD card inserted.

If I open the serial monitor, I only get garbage characters, no initialization or error messages.

Is my shield damaged, or am I doing something wrong?

supprot, 2017-04-15:
@irowebbn 
Hi,

The  ArduCAM_Shield_V2_Capture2SD.ino is used for our Shield_V2 board, you should use those exampleswhich is in MINI folder.

Let us know if you need more help.
Regards,
ArduCAM support team.

irowebbn, 2017-04-15:
Okay so I tried running ArduCAM_Mini_Capture2SD from /arducam/ArduCAM/examples/mini and I still get nothing on the indicator light or serial monitor. Nothing is being written to the memory card.

Perhaps I have selected the wrong hardware platform? I have the OV5642_MINI_5MP platform defined. Could you please tell me if this is correct?

```
//Step 1: select the hardware platform, only one at a time
//#define OV2640_MINI_2MP
#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS
```

Keep in mind I am using a 4 camera adapter board and a OV5642 Camera Shield (Rev 2.0 Sep 2016)

Thank you so much

supprot, 2017-04-16:
@irowebbn 
Hi,
Please attach us some picture about your phenomenon.
Let us know if yu need more help.
Regards.

irowebbn, 2017-05-05:
There was a hardware problem with the microSD card reader on the ArduCam mini platform. I purchased another one and the problem is gone.

closed 2017-05-05 by irowebbn