ArduCAM/Arduino #175: How do I connect the OV5642 to an Arduino Mega?

jwerbs1217, 2017-04-15:
Hi Arducam,

I have been testing out the OV5642 on my Arduino UNO and it has been working quite well. However, for my main project, I need to move it over to an Arduino MEGA , but when I switch to this Arduino it just gives me an SPI Interface Error. I am hooking it up to the exact same pins in the two arduinos but it is only working for the UNO. What should I do to fix this? I am using the LowPowerMode example for the arducam mini. 

Thanks!

supprot, 2017-04-16:
@jwerbs1217 
Hi,

What's your SPI CS ?Please attach us some pictures about your phenomenon.

Let us know if you need more help.
Regards,
ArduCAM support team.

jwerbs1217, 2017-04-18:
For anyone with this issue in the future, I figured it out. Connect MISO MOSI SCK to 50 51 52 respectively, CS to any digital pin (in the code this is designated as SPI_CS), VCC to 5V, GND to GND, and SDA and SCL to 20 and 21 respectively.

supprot, 2017-04-18:
@jwerbs1217 
Hi,

Please connect the SPI signal from ISCP.
Please refer to the picture we attached to you.
[image]
Let us know if you need more help.
Regards,
ArduCAM support team.

closed 2017-04-18 by jwerbs1217