ArduCAM/Arduino #177: ARDUCHIP_FRAMES not declared error when trying to compile VideoStream-example

joraldepat, 2017-04-21:
ArduCAM_Shield_V2_Video_Streaming:142: error: 'ARDUCHIP_FRAMES' was not declared in this scope

Could you please tell me how should this be declared?

supprot, 2017-04-21:
@joraldepat 
Hi,
Do you download our new ArduCAM library? We don't define the ARDUCHIP_FRAMES on 2MP.
If you are using OV2640 camera, just commment out the   myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);
We have do in the new library like this:

#if !(defined (OV2640_CAM)||defined (OV3640_CAM)) 
  myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
#endif
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

Please download our new ArduCAM library.
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Video_Streaming/ArduCAM_Shield_V2_Video_Streaming.ino

Let us know if you need more help.
Regards,
ArduCAM support team.

joraldepat, 2017-05-02:
> examples/Shield_V2/ArduCAM_Shield_V2_Video_Streaming/
> ArduCAM_Shield_V2_Video_Streaming.ino

That did it!  Thank you!