ArduCAM/Arduino #178: ARDUCAM OV2640......Really need your help..

hhoang3, 2017-04-23:
Can any one help us how to wiring the Arucam OV2640 with the UNO board, and what libraied we need to download ( step by step) because our team tried to wiring may ways, downloads alot libraries that people recommened, but still not work, the due date the coming but the camera part is not done. Please, we need your help:
RTC with sd card:
-----https://www.osepp.com/electronic-modules/breakout-boards/95-real-time-clock-microsd-breakout
UNO board:
----https://www.amazon.com/SunFounder-Arduino-ATMEGA328P-ATMEGA16U2-Version/dp/B00D9NEJ7U/ref=sr_1_14?ie=UTF8&qid=1492973776&sr=8-14&keywords=uno+board
Arducam OV2640:
----https://www.amazon.com/Arducam-Module-Megapixels-Arduino-Mega2560/dp/B012UXNDOY/ref=sr_1_1?ie=UTF8&qid=1492974369&sr=8-1&keywords=arducam
Arducam application host:
----http://www.arducam.com/downloads/demo/arducam_host/           (We tried the third and ninth)
Wiring: 
----https://www.google.com/search?q=arducam+wiring&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjZscmao7vTAhVC7IMKHQrtDmwQ_AUICSgC&biw=1532&bih=757#tbm=isch&q=arducam+wiring+to+uno+board&imgrc=aS9mken_lG1zLM:
Please, show us step by step to make it work....Thanks

supprot, 2017-04-24:
@hhoang3 
Hi,

Please refer to these below links to know more detail about how to use our ArduCAM library and hardware connection.
http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf

http://www.arducam.com/downloads/shields/ArduCAM_Mini_2MP_Camera_Shield_DS.pdf

Let us know if you need more help.
Regards,
ArduCAM support team.

hhoang3, 2017-04-24:
Sorry, Can I open the memorysaver by NotePad to uncomment or comment something....or do I need to download something as codeblock....?Thanks

supprot, 2017-04-24:
@hhoang3 
Hi,
You can do it through NotePad. 

Let us know if you need more help.
Regards,
ArduCAM support team.

hhoang3, 2017-04-24:
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))   . Do I need to comment this line also if I use the OV2460...However, after I uncomment, I can not save it... How can I do? Thanks ARducam team support..

supprot, 2017-04-24:
@hhoang3 
Hi,
You should enable the platform in the memorysaver.h
for example,if you are using MINI 2MP, you should enable
the #define OV2640_MINI_2MP and disable others.
the files in the library are just read and can not write.
You can save it in other path.

Let us know if you need more help.
Regards,
ArduCAM support teram.

hhoang3, 2017-04-24:
what do you mean when you said "You can save it in other path."...Can you show me how to do that.Thanks

supprot, 2017-04-24:
@hhoang3 
Hi,
Look at this picture,just choose OK and save it the path you choose.
[image]

Let us know if you need more heklp.
Regards,
ArduCAM support team.

hhoang3, 2017-04-24:
[image]
That how I got when I run the code....

hhoang3, 2017-04-24:
[image]
That some  of error I usually get when i run the code....However, I checked everything, but  nothing change

supprot, 2017-04-24:
@hhoang3 
Hi,
You should enable the platform in the memorysaver.h
for example,if you are using MINI 2MP, you should enable
the #define OV2640_MINI_2MP and disable others.
look at this picture:
[image]
Let us know if you need more help.
Regards,
ARduCAM support team.

hhoang3, 2017-04-24:
[image]
[image]
I did it, but I can not save it.....Sorry i may do not understand you something...please try to help me

hhoang3, 2017-04-24:
On laptop I  use NotePad to  enable the platform in the memorysaver.h

supprot, 2017-04-24:
@hhoang3 
Hi,
It is a so strange phenomenon,do you try reboot your computer?
Regards.

hhoang3, 2017-04-24:
why is strange>??

supprot, 2017-04-24:
@hhoang3 
Hi,
What do we feel strange is why you can not save the memorysaver.h using NotePad.
So we suggest you reboot your computer to try.
Regards.

hhoang3, 2017-04-24:
No...you told me above that"the files in the library are just read and can not write.
You can save it in other path."  So can anyone show me how to save another path on NotePad. Thanks

supprot, 2017-04-24:
@hhoang3 
Hi,
We are sorry to have kept you misunderstood our meaning.
What We mean is you can not change the examples in the library from Arduino IDE directly.
It will tell you some files are marked "read-only",so you need to save this sketch to another location.
Please look at the picture we send to you before.

Let us know if you need more help.
Regards,
ArduCAM support team.

Luddite-John, 2017-04-24:
There does not seem to be a simple diagram of how to connect the Arducam-mini-2Mp to an Arduino Uno, and a code sample that matches. 
I havew earchewd for over a week on github, arducam, you tube and google.

My plea is for a simple fritzing, and a code sample that matches, and a claear explanation of where any digression is required (such as do I need to program ARef). do I need to do anything but plug the mini into pins Digital 10 and up

hhoang3, 2017-04-24:
You mean after I enable the platform in the memorysaver.h, I need to save it to another location, so outside the libraries ( desktop) is ok? and that file need to make a same name is memorysaver...?

hhoang3, 2017-04-24:
@Luddite-John 
I connected the OV2640 to UNO follow by: 
https://www.google.com/search?q=arducam+to+UNO+board+circuit&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjy5sWguLzTAhVJ_IMKHa2nC2sQ_AUICCgB&biw=1920&bih=974#tbm=isch&q=arducam+circuits&imgdii=6EXyB2odMKtB9M:&imgrc=y-ENlpeRQlYSsM:
But after we run the example code, we could not preview the video on the HOST or take any picture to save to SD card(built in the RTC).

supprot, 2017-04-24:
@hhoang3 
No,
You just need to open the memorysaver.h file and enable your camera platform and save it instead of saving it in other location.
Then you should open your Arduion IDE file ->example -ArduCAM and choose the demo you need.
but you can not change the demo in the Arduino IDE directly. Because It will tell you some files are marked "read-only",so you need to save this sketch to another location.
Do you understand it ?

Let us know if you need more help.
Regards,
ArduCAM support team.

hhoang3, 2017-04-24:
I did enable your camera platform and save it.
I opened your Arduion IDE file ->example -ArduCAM and choose the demo I need, it run ok, but after I run the code I downloaded Windows host application, but the screen was all grey and I could not capture or preview live picture.
Then, I opened libraries->Arduino-master->Arducam->examples->mini->ArduCAM_Mini_Capture2SD and I run it , i got all error above
or "class Arducam has no members name get_bit"
      "ARDUCHIP_FRAME not declare in this scope"
Thanks

supprot, 2017-04-24:
@hhoang3 
Hi,
[image]

We don't define ARDUCHIP_FRAME  in the  MINI 2MP platform. 
Please try this below code:

// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
//
// This demo was made for Omnivision 2MP/5MP sensor.
// It will run the ArduCAM 2MP/5MP as a real 2MP/5MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM 2MP/5MP shield
// and use Arduino IDE 1.6.8 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define SD_CS 9
const int SPI_CS = 7;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, SPI_CS );
#else
  ArduCAM myCAM( OV5642, SPI_CS );
#endif

void myCAMSaveToSDFile(){
char str[8];
byte buf[256];
static int i = 0;
static int k = 0;
uint8_t temp = 0,temp_last=0;
uint32_t length = 0;
bool is_header = false;
File outFile;
//Flush the FIFO
myCAM.flush_fifo();
//Clear the capture done flag
myCAM.clear_fifo_flag();
//Start capture
myCAM.start_capture();
Serial.println(F("start Capture"));
while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
Serial.println(F("Capture Done."));  
length = myCAM.read_fifo_length();
Serial.print(F("The fifo length is :"));
Serial.println(length, DEC);
if (length >= MAX_FIFO_SIZE) //384K
{
  Serial.println(F("Over size."));
  return ;
}
if (length == 0 ) //0 kb
{
  Serial.println(F("Size is 0."));
  return ;
}
//Construct a file name
k = k + 1;
itoa(k, str, 10);
strcat(str, ".jpg");
//Open the new file
outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
if(!outFile){
  Serial.println(F("File open faild"));
  return;
}
myCAM.CS_LOW();
myCAM.set_fifo_burst();
while ( length-- )
{
  temp_last = temp;
  temp =  SPI.transfer(0x00);
  //Read JPEG data from FIFO
  if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
  {
    buf[i++] = temp;  //save the last  0XD9     
    //Write the remain bytes in the buffer
    myCAM.CS_HIGH();
    outFile.write(buf, i);    
    //Close the file
    outFile.close();
    Serial.println(F("Image save OK."));
    is_header = false;
    i = 0;
  }  
  if (is_header == true)
  { 
    //Write image data to buffer if not full
    if (i < 256)
    buf[i++] = temp;
    else
    {
      //Write 256 bytes image data to file
      myCAM.CS_HIGH();
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
    }        
  }
  else if ((temp == 0xD8) & (temp_last == 0xFF))
  {
    is_header = true;
    buf[i++] = temp_last;
    buf[i++] = temp;   
  } 
} 
}

void setup(){
uint8_t vid, pid;
uint8_t temp;
Wire.begin();
Serial.begin(115200);
Serial.println(F("ArduCAM Start!"));
//set the CS as an output:
pinMode(SPI_CS,OUTPUT);
// initialize SPI:
SPI.begin();
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  
  if (temp != 0x55){
    Serial.println(F("SPI interface Error!"));
    delay(1000);continue;
  }else{
    Serial.println(F("SPI interface OK."));break;
  }
}
//Initialize SD Card
while(!SD.begin(SD_CS)){
  Serial.println(F("SD Card Error!"));delay(1000);
}
Serial.println(F("SD Card detected."));

#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("Can't find OV2640 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV2640 detected."));break;
    } 
  }
#else
  while(1){
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if((vid != 0x56) || (pid != 0x42)){
      Serial.println(F("Can't find OV5642 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV5642 detected."));break;
    } 
  }
#endif
myCAM.set_format(JPEG);
myCAM.InitCAM();
#if defined (OV2640_MINI_2MP)
  myCAM.OV2640_set_JPEG_size(OV2640_320x240);
#else
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
#endif
delay(1000);
}
void loop(){
myCAMSaveToSDFile();
delay(5000);
}

Regards,
ArduCAM support team.

Luddite-John, 2017-04-24:
Hi hhoang3.

Thanks for the link to the circuit.  I connected it and still had a problem.

I looked over a lot of other information I had, and I found a comment to move SPI Slave Chip Select input (CS) from Digital Pin 10 to to Digital pin 7, and then it worked for me (I am using example>Mini>Arducam_Mini_LowPowerMode)

I suggest you try moving Pin 1 on Mini-2mp board from digital pin 10 to digital pin 7 on the Uno  and rerun it from the start configuration.

hhoang3, 2017-04-25:
I copied yoour code above and run it, I save it in desktop.
I still got same error : 
//#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
I comment this line , but I got error: 
[image]
I defined the camera in the memorysaver already:
#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_

//Only when using raspberry,enable it
//#define RASPBERRY_PI

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
//#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
    #define OV2640_CAM
	//#define OV3640_CAM
	 //#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM//
//#endif 

#endif	//_MEMORYSAVER_

hhoang3, 2017-04-25:
for the memmorysaver, after I changed it, I clicked save but could not...

hhoang3, 2017-04-25:
Please someone help us?

hhoang3, 2017-04-25:
[image]
That how I get all the time...

UCTRONICS, 2017-04-26:
@hhoang3 
Hi,
Please use our Host_V2.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app
Let us know if you need more help.
Regards,
ArduCAM support team.

hhoang3, 2017-04-26:
@Luddite-John 
I tried your way but can not run the code...I get the error like this 
[image]

Luddite-John, 2017-04-26:
Hi,
There are a number of things that need to be right for the AMC (Arduino Mini-2mp Camera - OV2640) to work.

1. Did you resolve the issue about not being able to solve how to save the memsaver.h changes, if you cannot save it after change, are you on a company/ business Network (If so ask Network administrator to confirm you ccan save the file memsaver.h to the location you need to save it to).
.
2 . I suggest you use the wiring as per this table:
Shield    UNO
S1     -      D7
S2     -      D11
S3     -      D12
S4     -      D13
S5     -      Gnd anywhere
S6     -      5v (do not use A Ref)
S7     -      (Would be D16 but pin not marked)
S8     -      (Would be D17 but pin not marked)  
.
3. I would reinstall the Arducam Library (where you keep your sketches), this will remove any errors in code changes
.
4 Modify memsaver.h in the arducam library (you saved in 3 above ) to define only the OV2640 as explained earlier and then save the changes over the top of the old one.
.
5. Use the example "Arducam_Mini_LowPowerMode" from the Mini library of the ARDUCAM library (you will not have to change the CS pin if you wired as per 2 above
.
6. Then check the result with the arducam.exe program

Hope this helps

hhoang3, 2017-04-26:
what do you mean " a company/ business Network (If so ask Network administrator to confirm you ccan save the file memsaver.h to the location you need to save it to)."
.Can you check for me is I did right?if not, what I need to comment or uncomment . Thanks

#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_

//Only when using raspberry,enable it
//#define RASPBERRY_PI

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
    //#define OV2640_CAM
	//#define OV3640_CAM
	 #define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 

#endif	//_MEMORYSAVER_

Luddite-John, 2017-04-26:
If you are at work, and your are on a network, then you may not have the rights or permissions that allow you to save the file in a certain location.

If you have a Mini 2 MP(OV2640), then the memsaver.h should read:

#ifndef MEMORYSAVER
#define MEMORYSAVER

//Only when using raspberry,enable it
//#define RASPBERRY_PI

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC
//#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
//#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
//#define OV7660_CAM
//#define OV7725_CAM
//#define OV7670_CAM
//#define OV7675_CAM
//#define OV2640_CAM
//#define OV3640_CAM
//#define OV5642_CAM
//#define OV5640_CAM

//#define MT9D111A_CAM
//#define MT9D111B_CAM
//#define MT9M112_CAM
//#define MT9V111_CAM	
//#define MT9M001_CAM	
//#define MT9T112_CAM
//#define MT9D112_CAM

#endif

#endif //MEMORYSAVER

hhoang3, 2017-04-26:
I am not at work, I am using my laptop at home,but still can not save..
Quick question: in step 2, do we need to uncomment the line //#define OV2640_CAM
and comment one of the #endif at the end. Thanks @Luddite-John

Luddite-John, 2017-04-26:
Hi,
That is the same Mini 2MP camera I just worked on and got to work on the UNO over the weekend.

Yes, uncomment //#define OV2640_CAM  and comment out the first  #endif.

For your camera the the file reads like (ignoring all of the commented out bits):
#ifndef MEMORYSAVER
#define MEMORYSAVER
#define OV2640_MINI_2MP
#endif //MEMORYSAVER

What this file says is:  
- if "memorysaver" not defined, 
- then define "memorysaver", 
- then it defines "OV2640_Mini_2MP", 
- then it ends the if "memorysaver" not defined,

So this whole file simply is saying where you see memorysaver, substitute the words with OV2640_Mini_2MP.

If the file does not save back to the same location or it has an error saving, please include screen shot of the error.

Then restart the Arduino IDE and run the example for low power again..

Once you have this saved and is right then we can look at the errors.

hhoang3, 2017-04-26:
@Luddite-John 
After I click to save they asked me do I want to replace the file memorysaver. I click ok and I got
[image]
Or do we need to save with other name, or delete the old one....
I can not save it in my school also

hhoang3, 2017-04-26:
@supprot @UCTRONICS 
Can you help me about that saving for memorysaver?
I alway have error when I run the code"#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file"
Thanks

Luddite-John, 2017-04-27:
Hi Hhoang,

Your issue isn't the memorysaver.h file, or an Arducam problem, instead it is a problem with saving a file, and that is not what this thread is designed to address.

I have 2 suggestions:
A. Make the changes to the  memorysaver.h file, and then try to save the file to your desktop; then using windows file explorer copy the file (from your desktop) over the top of the memorysaver.h file (in the library folder), and if it does not let you, then your problem is with the rights and permissions you have to that directory. 
.
B. Use file Explorer (or similar), to look at the libraries directory and the files in the directory and see what permissions you have and what the file rights are set at, then google how to change those permissions/ rights.

Until you can resolve this issue, there isn't much that can be done to make the Arducam work.

hhoang3, 2017-04-27:
I did it, I could change the memorysaver.h file and saved it immediately. I will try to run the code tonight...I hope everything work fine, if not, i need your guys have again.Thanks

hhoang3, 2017-04-27:
I did the memorysaver done, but I still get the error like this again:
[image]
Please help me!!!Thanks

hhoang3, 2017-04-27:
I also get this one if I use others code...
[image]

hhoang3, 2017-04-27:
C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode\ArduCAM\examples\mini\ArduCAM_Mini_2MP_TimeElapse2SD\ArduCAM_Mini_2MP_TimeElapse2SD.ino: In function 'void loop()':

ArduCAM_Mini_2MP_TimeElapse2SD:112: error: 'class ArduCAM' has no member named 'get_bit'

   while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

                 ^

ArduCAM_Mini_2MP_TimeElapse2SD:130: error: 'class ArduCAM' has no member named 'CS_LOW'

   myCAM.CS_LOW();

         ^

ArduCAM_Mini_2MP_TimeElapse2SD:131: error: 'class ArduCAM' has no member named 'set_fifo_burst'

   myCAM.set_fifo_burst();

         ^

ArduCAM_Mini_2MP_TimeElapse2SD:146: error: 'class ArduCAM' has no member named 'CS_HIGH'

       myCAM.CS_HIGH();

             ^

ArduCAM_Mini_2MP_TimeElapse2SD:150: error: 'class ArduCAM' has no member named 'CS_LOW'

       myCAM.CS_LOW();

             ^

ArduCAM_Mini_2MP_TimeElapse2SD:151: error: 'class ArduCAM' has no member named 'set_fifo_burst'

       myCAM.set_fifo_burst();

             ^

ArduCAM_Mini_2MP_TimeElapse2SD:157: error: 'class ArduCAM' has no member named 'CS_HIGH'

     myCAM.CS_HIGH();

           ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\ERIC\Documents\Arduino\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD-master
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'class ArduCAM' has no member named 'get_bit'

Invalid library found in C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode: C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\Arduino-master: C:\Program Files (x86)\Arduino\libraries\Arduino-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\example: C:\Program Files (x86)\Arduino\libraries\example
Invalid library found in C:\Program Files (x86)\Arduino\libraries\examples: C:\Program Files (x86)\Arduino\libraries\examples
Invalid library found in C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master: C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\library: C:\Program Files (x86)\Arduino\libraries\library
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MotionClock: C:\Program Files (x86)\Arduino\libraries\MotionClock
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MotionClock-master: C:\Program Files (x86)\Arduino\libraries\MotionClock-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\NewMotionClock: C:\Program Files (x86)\Arduino\libraries\NewMotionClock
Invalid library found in C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode: C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\Arduino-master: C:\Program Files (x86)\Arduino\libraries\Arduino-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\example: C:\Program Files (x86)\Arduino\libraries\example
Invalid library found in C:\Program Files (x86)\Arduino\libraries\examples: C:\Program Files (x86)\Arduino\libraries\examples
Invalid library found in C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master: C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\library: C:\Program Files (x86)\Arduino\libraries\library
[... 67 lines cut ...]
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MotionClock-master: C:\Program Files (x86)\Arduino\libraries\MotionClock-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\NewMotionClock: C:\Program Files (x86)\Arduino\libraries\NewMotionClock
Invalid library found in C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode: C:\Program Files (x86)\Arduino\libraries\arducam-fix-low-power-mode
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_v1.0.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0: C:\Program Files (x86)\Arduino\libraries\ArduCAM_V1.3.0
Invalid library found in C:\Program Files (x86)\Arduino\libraries\Arduino-master: C:\Program Files (x86)\Arduino\libraries\Arduino-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\example: C:\Program Files (x86)\Arduino\libraries\example
Invalid library found in C:\Program Files (x86)\Arduino\libraries\examples: C:\Program Files (x86)\Arduino\libraries\examples
Invalid library found in C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master: C:\Program Files (x86)\Arduino\libraries\LCD-KeyPad-Shield-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\library: C:\Program Files (x86)\Arduino\libraries\library
[... cut ...]
Invalid library found in C:\Program Files (x86)\Arduino\libraries\library: C:\Program Files (x86)\Arduino\libraries\library
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MotionClock: C:\Program Files (x86)\Arduino\libraries\MotionClock
Invalid library found in C:\Program Files (x86)\Arduino\libraries\MotionClock-master: C:\Program Files (x86)\Arduino\libraries\MotionClock-master
Invalid library found in C:\Program Files (x86)\Arduino\libraries\NewMotionClock: C:\Program Files (x86)\Arduino\libraries\NewMotionClock
Please help our team...Thanks

supprot, 2017-04-27:
@hhoang3 

Hi,
Step 1:
Install your Arduino IDE.
https://www.arduino.cc/en/Guide/ArduinoUno
Setp 2:
Download our ArduCAM library from https://github.com/ArduCAM/Arduino.git
Step3:
Put our ArduCAM library to your ..\arduino\libraries.
Step4:
Configure the memorysaver.h file which is under the ArduCAM folder  in order to run your camera.
Step5:
Open your Arduino IDE choose  File->Examples->ArduCAM ..
[image]
[image]
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM supoort team.

hhoang3, 2017-04-27:
I did all of them...but the camera still not work....so worry right now...

supprot, 2017-04-27:
@hhoang3 
Hi,
From the pictures you have attached, you still can't compile the code successfully.
If you have done those steps we send to you right, we don't think this phenomenon is exist.
Please retry from step one.

Let us know if you need more help.
Regards,
ArduCAM support team.

hhoang3, 2017-04-27:
Can Anyone help me, because I do not know why the Arducam libraries that link about not inclide in the libraries . I downloaded the Arducam zip copied that in the libraries of Arduino folder.
[image]
[image]

hhoang3, 2017-04-27:
[image]

supprot, 2017-04-28:
@hhoang3 
Hi,
You should put ArduCAM in the library.
Please check the previous picture we attached for you.

Regards,
ArduCAM support team.

hhoang3, 2017-04-28:
I tried but I could not. I usually downloaded the libraries from github and copy that files into the libraries of Arduino. However, I did same thing but the libraries does not appear .Can you show me how to put it in?Thanks

hhoang3, 2017-04-29:
My camera is working fine right now.Thanks you guys. However, can you help me how to save the picture to the micro SD after taking....
I using: 
https://www.osepp.com/electronic-modules/breakout-boards/95-real-time-clock-microsd-breakout
I hope you guys can help.Thanks

closed 2017-04-30 by hhoang3

supprot, 2017-05-02:
@hhoang3 
Hi,
Glad to hear you have solved your problem.
Please refer to our demo :
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_Capture2SD

Regards,
ArduCAM support team,

blitobyte, 2019-04-05:
Hello,
I've been working on a project that includes the use of ten sensors with Arducams mini 2MP OV2640. Having my prototype ready I proceeded to buy the rest of the cameras. When I got them I discovered that the first two I tested wouldn't take a proper image. What I got was a black image with a tiny little white-bordered square in the middle (on Chrome). I then looked more carefully at the cameras and discovered that they weren't identical to the one I already had. Although the pins were the same and my Arduino program didn't complain in debug  mode, the chips on the cameras were different.
I looked into the files and realised the first byte in the images from the new cameras wasn't the same. In the good images the first two bytes were FF D8 whereas in the bad images the FF was missing. Sometimes, even the new image files included a 00 instead of the FF before the D8.  With a small python script I rewrote the bad images prepending the FF and... voila! The images were ok.
Now, the question is, why is this happening? What is the difference between the two Arducam models? Should I do some sort of set-up with this new cameras?
By the way, the prototype works on a ESP8266, not an Arduino.

I'm sending you some photos of both cameras and an example of the image I get.

Thanks a lot,

Pablo

[image]
[image]
[image]

tortuga37, 2019-12-12:
You have to change the baud rate in the sample code to a smaller rate.  The Arduino Uno board can not handle the fastest baud rate.