ArduCAM/Arduino #179: ARDUCAM Mini-2mp (OV2640) connection with Arduino Uno R3

Luddite-John, 2017-04-24:
Does anyone have a simple diagram (and expereince ) of how to connect the Arducam-Mini-2Mp to an Arduino Uno, and a code sample that matches.
After a number of false starts, I have connected the Mini-2Mp to the Arduino Uno R3 as follows:
Mini-2Mp Pin 1 - Uno D10
Mini-2Mp Pin 2 - Uno D11
Mini-2Mp Pin 3 - Uno D12
Mini-2Mp Pin 4 - Uno D13
Mini-2Mp Pin 5 - Uno  Digital Gnd pin
Mini-2Mp Pin 6 - Uno Power connector pin (5v)
Mini-2Mp Pin 7 - Uno 3 pins above D13
Mini-2Mp Pin 8 - Uno 4 pins above D13
I have installed Arducam in my library and commented out everything except OV2640 in memorysaver.h and then ran all examples for File>Examples> Arducam> mini), then used Arducam _v1.0.exe to control and review the output. 
Adjusting the baud rate to either 115200 or 921600 I get a readable output.
For example runnning the sketch "Arducam_Mini_2MP_LowPowerMode", gives an output in Arducam_V1.0.exe of:
Port Opened.
ACK CMD ArduCAM Start!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error! 

I get different errors for each .ino (example/ sketch) that I run, but it never has produced an image.Port Opened.
ACK CMD ArduCAM Start!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!
ACK CMD SPI interface Error!

Any help would be greatly appreciated

supprot, 2017-04-24:
@Luddite-John 
Hi,

In our demo ,we define SPI CS 7 instead of 10.
Please connect the pin 1 to uno 7 instead of 10.

Let us know if you need more help.
Regards,
ArduCAM suport team.

Luddite-John, 2017-04-24:
Thank you for responding.

I came across the same message where you had responded to someone else with a similar problem.

I used the UNO to simplify the learning path, as i am trying to interface the Arducam min 2mp to a Mega 2650, and I thought it would be easier to use the Uno as there was a lot more data .

I have now connected the Mini 2mp to the Arduino Mega (using exactly the same pin configuration i used on the UNo), and once again it has the SPI error "ACK CMD SPI interface Error!" 

I tried connecting CS (pin 1 on mini 2mp) to the Digital pin 10 and also to Digital Pin 7, but neither worked.

Is there other differences/ programming required to use the Mega in polace of the UNO?

Thanks

supprot, 2017-04-24:
@Luddite-John 
Hi,
On Mega2560 platform, we suggest you use SPI signals from the ICSP.
Look at the picture we attached to you:
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

Luddite-John, 2017-04-24:
Thanks Arducam Support Team.

I have made the wiring changes as per your suggestion, 
I think the table for the Arducam mini-2mp shield looks like this:

Pin TYPE	        Type                  Description                                    UNO-R3       Mega - 2560
1	CS	        Input	         SPI slave chip select input	        D7	             D10
2	MOSI	Input	         SPI master output slave input	D11              ICSP4
3	MISO	Output	         SPI master input slave output	D12              ICSP1
4	SCLK	Input	         SPI serial clock	                        D13	             ICSP3
5	GND	Ground	         Power ground	                       Gnd	             Gnd
6	+5V	        POWER	         5V Power supply	                        VCC	             VCC
7	SDA 	Bi-directional	Two-Wire Serial Interface Data I/O	D16  	     D20
8	SCL	       Input	                Two-Wire Serial Interface Clock	D17	             D21

Are there any code changes or pin reallocation required (such as previously in ArduCAM_Mini_LowPowerMode sketch you changed  your code for the Mini 2mp uses D7 instead of  D10).

What if any code changes are required to make the camera work in the configuration for the Mega

Thanks

supprot, 2017-04-24:
@Luddite-John 
Hi,
About SPI CS,it's up to you, we just provide a demo,you should not change others except CS.
If you set the CS 7 in the code,you shoud connect SPI CS to 7.

Let us know if you need more help.
Regards,
ArduCAM support team.

Luddite-John, 2017-04-24:
Thank you.

I understand this, so I have change SPI Cs harware to pi 7 and run it no work, says "CMD SPI Interface Error".
So I change hardware to digital pin 10 and change SPI Cs to pin 10 and run again, still says "CMD SPI Interface Error".

So problem not using pin 7 or pin 10.  Do you know what other mistake can cause error message "CMD SPI Interface Error".

supprot, 2017-04-24:
@Luddite-John 
Hi, 
It also maybe a hardware connection problem.
You can check it by oscilloscope.

Let us know if you need more help.
Regards,
ArduCAM support team.

Luddite-John, 2017-04-24:
Hi,

Yes I need help.

Interesting that you think it might be a hardware problem.

So I switched everything back and ran the mini 2mp shield on the Arduino Uno again, and it worked fine.

When I connect ujp min 2mp to Arduino Mega again it does not work.

So no, not a hardware problem.

Please you can help by writing what each of the 8 pins on the Mini 2MP shield must go to on the Arduino Mega, to run the sample code you have provided, as what you have provided to date does not work.

If you do not have a list, then maybe a picture with a working one, or review the list I provided above and make corrections as needed.

Thank you

supprot, 2017-04-24:
@Luddite-John 
Hi,
Very glad to hear your camera can work fine on UNO.
Please refer to those pictures we attached for you.
We use our adapter in order to make multi cameras work together.
We chose 5 and 7 as SPI CS.
[image]
[image]

[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

Luddite-John, 2017-04-25:
Thanks for your timely responses and your continued efforts to resolve (successfully), my original issue.

I went back to my original question and I now believe that this issue thread should now be closed (I will raise a separate issue on connecting the Mini-2mp shield to the Mega), as the advice provided has resulted in a working camera using the Mini-2MP Shield and an Arduino Uno R3

For anyone reading this thread who also has this issue (Arducam Mini -2MP camera not working on UNO), I make this reccm: 

-  Connect the camera shield to the UNO using 15cm male to female jumper leads. (I am not a big fan of bending legs on an experimental piece of kit).

- Next,  connect all pins directly into the digital pins (pin2 on the shield to D11, d3 to pin 12 etc.

- Except VCC  (pin 6 on the Shield), which you connect to the 5v on the analogue strip (not ARef [leave this empty]) and also connect CS (pin 1 on the shield) to Digital pin 7 and 

- Run the "ArduCAM_Mini_LowPowerMode.ino" from the examples (after you install ARDUCAM into your library)

closed 2017-04-25 by Luddite-John

robkwan, 2017-05-09:
Hello!

I am totally new to this ArduCAM_Mini development 
and I just got my setup today with:-

1. Arduino ESP 8266 UNO;
2. OV2640 Camera Shield for Arduino;
&
3. ArduCAM 4 Cameras Adapter Board

And I installed Arduino IDE 1.8.2.

I have no problem to run the basic examples with 1 & 2 only.

However, when I try to add the 4 camera adapter board in-between, nothing works!

I tried to download the ArduCAM_Mini_4CAM_Capture2SD but it would crash immediately after bootup.

I tried to follow the document:
Multi-Camera Adapter Board User Guide Rev 1.1, Oct 2015
and it mentions about ArduCAM_Mini_2MP_4Cams_TimeElapse2SD_LowPower.ino

But I find that I cannot use it either even if I can find it from github
because the codebase is so different from 2015 to now?!

Therefore, I would like to ask if there would be another example .ino which can allow me to plug in the Camera to the Adaptor board and run ok please.
Initially, I just want to have a good stand for the one and only one camera first
but I may buy the other 3 later if it really works...!

Any suggestion please?

Thanks & Best Regards,
Robert

UCTRONICS, 2017-05-09:
@robkwan 
Hi,

About ESP8266 UNO, please refer to this document:
http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf
We have update our codes, so it will exist some differents.

The ArduCAM Mini codes are used for Arduino UNO board instead of ESP8266 UNO board.
 
Let us know if you need more help.
Regards,
ArduCAM support team.

robkwan, 2017-05-09:
Hello!

Thanks for your prompt reply!

However, as I said before, I have no problem to use the ESP8266 UNO board
with the Camera.

So is there any reference example/s which can show me how to use the 4
Camera Adapter board with ESP8266 UNO board please?!
Or it is not possible to use the Adapter board with ESP8266 UNO board pls?!

Thanks & BR,
Robert

UCTRONICS, 2017-05-09:
@robkwan 
Hi, At ESP8266 platform, we don't have enough pin sources,
so At present our ESP8266 can only support two cameras work together.
Please refer to this code we atached for you,but you can't use our adapter directly
for ESP8266, you can connect the other camera by dupont line.

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini 2MP camera, and can run on any Arduino platform.
//
// This demo was made for ArduCAM Mini Camera.
// It needs to be used in combination with PC software.It can take 4 photos at the same time with 4 cameras.
// The demo sketch will do the following tasks:
// 1. Set the 4 cameras to JPEG output mode.
// 2. Read data from Serial port and deal with it
// 3. If receive 0x00-0x08,the resolution will be changed.
// 4. If receive 0x15,cameras will capture and buffer the image to FIFO. 
// 5. Check the CAP_DONE_MASK bit and write datas to Serial port.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM Minicamera
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

// set pin 4,5,6,7 as the slave select for SPI:
const int CS1 = 16;
const int CS2 = 0;

//the falgs of camera modules
bool cam1=true,cam2=true,cam3=true,cam4=true;
//the flag of JEPG data header
bool is_header;
//the falg data of 4 cameras' data
byte flag[5]={0xFF,0xAA,0x01,0xFF,0x55};
int count = 0;
#if defined (OV2640_MINI_2MP)
ArduCAM myCAM1(OV2640, CS1);
ArduCAM myCAM2(OV2640, CS2);
#else
ArduCAM myCAM1(OV5642, CS1);
ArduCAM myCAM2(OV5642, CS2);
#endif

void setup() {
// put your setup code here, to run once:
uint8_t vid,pid;
uint8_t temp;
Wire.begin(); 
Serial.begin(921600);
Serial.println("ArduCAM Start!"); 
// set the CS output:
pinMode(CS1, OUTPUT);
pinMode(CS2, OUTPUT);

// initialize SPI:
SPI.begin(); 
while(1){
  //Check if the 2 ArduCAM Mini 2MP Cameras' SPI bus is OK
  myCAM1.write_reg(ARDUCHIP_TEST1, 0x55);
[... 508 lines cut ...]
  temp =  SPI.transfer(0x00);//read a byte from spi
  if(is_header == true)
  {
    Serial.write(temp);
  }
  else if((temp == 0xD8) & (temp_last == 0xFF))
  {
    Serial.println(F("ACK IMG"));
    is_header = true;
    Serial.write(temp_last);
    Serial.write(temp);
  }
  if( (temp == 0xD9) && (temp_last == 0xFF) )
  break;
  delayMicroseconds(15);
}
myCAM.CS_HIGH();
is_header = false;
//Clear the capture done flag 
myCAM.clear_fifo_flag();
}

Let us now if you need more help.
Regards,
ArduCAM support team.

robkwan, 2017-05-09:
Hello!

Thanks for the information!

Two is enough for the moment as I mainly want the stand support from the
adapter board
but I don't really understand what you meant by:-

but you can't use our adapter directly
for ESP8266, you can connect the other camera by dupont line.

If so, it means that I still can't get the camera standing right like as on
adapter board?!

Any other adapter board?
Or suggestion please?!

Thanks & Best Regards,
Robert

robkwan, 2017-05-09:
Sorry that I might not be giving the right information to question!

In fact, I did use the dupont lines to connect the Camera to the ESP8266
UNO board
and all the samples are working fine.

However, if I plug the camera into the middle slot as attached picture from
the document, it would not work.

Any idea please?!

Thanks & BR,
Robert

UCTRONICS, 2017-05-10:
@robkwan 
Hi,
please attach us your connection picture.
Regards,

robkwan, 2017-05-10:
Hello!

Here I attach two photos:-

1. One working with DupontCables but I didn't really re-connect after I
unplugged it.

2. One not working when I just plug the camera module into the middle pin
lot as I captured from the document yesterday.

Thanks & Best Regards,
Robert

UCTRONICS, 2017-05-10:
@robkwan 
We haven't found  your hardware connection picture.Please attach it again.

robkwan, 2017-05-10:
Does it work this time please in the attachments?!

Thanks & BR,
Robert

---------- Forwarded message ----------
From: Robert Kwan <robert.kwan@gmail.com>
Date: Tue, May 9, 2017 at 6:22 PM
Subject: Re: [ArduCAM/Arduino] ARDUCAM Mini-2mp (OV2640) connection with
Arduino Uno R3 (#179)
To: ArduCAM/Arduino <
reply@reply.github.com
>
Cc: ArduCAM/Arduino <Arduino@noreply.github.com>, Mention <
mention@noreply.github.com>

Hello!

Here I attach two photos:-

1. One working with DupontCables but I didn't really re-connect after I
unplugged it.

2. One not working when I just plug the camera module into the middle pin
lot as I captured from the document yesterday.

Thanks & Best Regards,
Robert

robkwan, 2017-05-10:
​

supprot, 2017-05-10:
@robkwan 
[image]
We can't see your picture....

robkwan, 2017-05-10:
[image]

Above is the one not working...!!

[image]

However, using dupont cable wires, no problem.
And the photo was taken after I unplugged the pins only.

Thanks & BR,
Robert

robkwan, 2017-05-10:
I also created another new issue for my issue as I learned more about the ArduCAM module please.

https://github.com/ArduCAM/Arduino/issues/190

Thanks & BR,
Robert

robkwan, 2017-05-10:
Hello!

I just found that my Arducam ESP8266 UNO V2 board keeps resetting once I connect the micro-USB to my Windows 10 laptop now.  Because I turned on the speaker and the beep sound of driver issue.
And so the COM7 port keeps resetting and I cannot even use the Arduino IDE to reflash the ArduCam_ESP8266_Capture.ino anymore.

Not sure if that would explain why I said that the setup would not work when I plugged in the camera module?!

Any way to reset the Arducam ESP8266 UNO V2 board to normal state please?!

Thanks & Best Regards,
Robert

Vazrul, 2017-07-07:
please , how to connect omnivision camera ov2640 to arduino , and the pin omvision camera ov2640 have 20 pin ???

tharangamuditha, 2017-09-06:
Can anybody please give a detailed information or internet sources on programming Arducam ESP8266 UNO with ArduCAM Mini 2MP camera on making a live video streaming into a web page?
Thanks & Best regards,
Tharanga.

supprot, 2017-09-06:
@tharangamuditha
Hi,
Please refer to this link:
http://www.arducam.com/knowledge-base/esp8266-uno-tutorial/

tharangamuditha, 2017-09-06:
Thank you for quick reply. But I already gone through it and got the
outputs for image. But I could not get the video streaming.
when the video.html file is opened, is a live video streaming takes place?

Thanks and Best regards,
Tharanga.

tharangamuditha, 2017-09-06:
Thank you for quick reply. But I already gone through it and got the outputs for image. But I could not get the video streaming. when the video.html file is opened, is a live video streaming takes place? 
Thanks and Best regards,
Tharanga.
--

supprot, 2017-09-06:
@tharangamuditha
Hi,
You should choose one resolution.Then the video will be send to the web.
Please try it again.
Regards.

sohombrg, 2018-07-24:
Hi I am have successfully connected Arducam mini module with ESP8266 nodemcu and it works fine. But now I have only 1 digital pin free for other purposes. So I wanted to know if we can modify the default wiring configuration. If so, can any one tell me how to do it.

UCTRONICS, 2018-07-24:
@sohombrg 
ESP8266 nodemcu doesn't have enough pin source. You can use the ESP32 which has enough gpio source.
http://www.uctronics.com/arducam-esp32-uno-board-for-arducam-mini-camera-module-compatible-with-arduino-uno-r3.html