ArduCAM/Arduino #181: Struggles with Arducam Mini 2MP

JarkJVC, 2017-04-26:
Hi,

I want to use the arducam mini 2mp to take a picture after receiving a voltage signal, but I can't even a picture at the moment. I think I might have wired it wrong, but I am not sure. I am using an Arduino Uno.

It's hard to make out, but here's the error message I'm getting

[image]

The SPI errors make me think I've done something wrong. 
The board is wired as follows 

SCL - SCL
SDA - SDA
VCC - 3.3V
GND - GND 
SCK - 7
MISC - 6
MOSI - 5
CS - 4
[image]
[image]
[image]

Any help at all would be greatly appreciated. Thanks.

Luddite-John, 2017-04-26:
Hi,

There are a number of things that need to be right for the AMC (Arduino Mini-2mp Camera - OV2640) to work.

A. For your camera to work, the the memorysaver.h file needs to be amended so that file reads like (ignoring all of the commented out bits):
#ifndef MEMORYSAVER
#define MEMORYSAVER
#define OV2640_MINI_2MP
#endif //MEMORYSAVER
.
B . I suggest you use the wiring as per this table:
Shield - UNO
S1 - D7
S2 - D11
S3 - D12
S4 - D13
S5 - Gnd anywhere
S6 - 5v (do not use A Ref)
S7 - (Would be D16 but pin not marked)
S8 - (Would be D17 but pin not marked)
.
C. Run the example from the mini examples using example>Mini>Arducam_Mini_LowPowerMode (this sketch uses pin 7 as CSas per the table above
.
D. Then you can make the changes needed as you try the different programs (any examples with a 4 in the title, are expecting you to be running the 4 Camera shield (and looking at your pictures, you are not), so I suggest avoid them.

JarkJVC, 2017-04-26:
Thanks, that fixed it. Would you happen to know if it's possible to take pictures automatically? Basically, I have a circuit that send out a 5V signal under certain circumstances. When those conditions are met, I want to the camera to a picture automatically. Also, I'm trying to figure out how the continuous shots mode works. I doesn't seen to do anything at the moment.

Luddite-John, 2017-04-27:
Hi JarkJVC,

Glad to help, and i am happy that you got it to work.

I find the operation is very touchy and suffers with poor connections (using jumping wires), I also note that poorly designed shields and using long wires to connect, provide less stability and introduces the odd error in the pictures taken (I am going to solder it all together once I am certain it works the way I want).

You are at the same point as me, and that is now the camera is set up but to operate it requires the use of the Arducam_V1.exe (or similar), so I dont have a quick way for you to get it to work on a single event.

My next task is to look at the source code for Arducam.exe  and try to gain an understanding of exactly what happens when you press the capture button (I assume there is a set up (as I see in the serial window it changes the set up to a certain mode) and I imagine there is a specific command line that passes all of the needed parameters across to make it happen). 

If you haven't already got them. then  I suggest you get the users and programming guides etc (there are 3 PDFs), that explain in broad terms how to program it.

I would love to share in what you learn, as I intend to use Python scripts to enact everything needed (by interfacing a RPiZ-W, to do all the smarts), then use an external trigger of 5volts to take the picture.

allanah1, 2019-02-25:
Hey

Were any of you able to trigger the camera automatically without the software?

UCTRONICS, 2019-02-27:
@allanah1 
Hi,
We have added an external trigger to start capture demo. You can get it from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_PIR_Trig_Capture2SD.ino/ArduCAM_Mini_PIR_Trig_Capture2SD.ino
In the demo,  we use a PIR sensor to detect the rising edge then start a snapshot.