ArduCAM/Arduino #185: ACK CMD Can't find OV2640 module!

Korzonco, 2017-05-03:
Hi,

This is what I have done so far..
1) changing the memorysaver.h for OV2640 only
2) using the ArduCAM_Mini_LowPowerMode.ino code from your site
3) changing "const int SPI_CS = 7;" to "const int SPI_CS = 10;"
4) Connecting the hardware as followed:
Mini-2Mp to Arduino Uno R3:
Mini-2Mp Pin 1 - Uno D10
Mini-2Mp Pin 2 - Uno D11
Mini-2Mp Pin 3 - Uno D12
Mini-2Mp Pin 4 - Uno D13
Mini-2Mp Pin 5 - Uno Digital Gnd pin
Mini-2Mp Pin 6 - Uno Power connector pin (5v)
Mini-2Mp Pin 7 - Uno 3 pins above D13
Mini-2Mp Pin 8 - Uno 4 pins above D13
5) running the code and it gives me this:
ACK CMD SPI interface OK.
ACK CMD Can't find OV2640 module!

I need some help.

UCTRONICS, 2017-05-03:
@Korzonco 
Hi,
Mini-2Mp Pin 7 - Uno  pin4 (SDA)
Mini-2Mp Pin 8 - Uno  pin5 (SCL)
Regards,

Korzonco, 2017-05-03:
Hi,
Tried, no luck.
Regards

UCTRONICS, 2017-05-03:
@Korzonco 
Hi,

If your I2C bus is OK,you can try to print the value of  pid and vid to check.

Regards,