ArduCAM/Arduino #186: Can't find OV2640 module! pid: 254

Lornioiz, 2017-05-03:
Hello!,

According to the pcb, I own a "Arducam-Mini-2MP" (bought in october 2016). In the small box that contained the camera there was this text: "Mini Arducam module Camera 2 MP OV2460".

If I execute a sketch relative to the OV2640, i.e. "ArduCAM_ESP8266_OV2460_Capture", I get ithe following error:
`Can't find OV2640 module! pid: 254
`
The I printed out the vid and pid values, finding these values:

vid = 239
pid = 254

I don't understand why these value are so different than those expected.
How can i fix the problem?

Thanks!

UCTRONICS, 2017-05-04:
@Lornioiz 
Hi,

We have update our ESP8266 UNO,please delete the old version and download the new version.
http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json,http://arduino.esp8266.com/stable/package_esp8266com_index.json
When you use 2MP, you should enable OV2640_MINI_2MP and disable others firstly.

Let us know if you need more help.
Regards.

Lornioiz, 2017-05-04:
Thank you for your answer.
I have a generic esp8266 module, Should I use the the ESP8266_UNo nonetheless?

Thanks!

supprot, 2017-05-05:
@Lornioiz 
Hi,

Yes, It should be OK.but you should pay attention to the signal to redefine.

Let us know if you need more help.
Regards,
ARduCAM support team.

Lornioiz, 2017-05-08:
Hello,

sorry to bother you again.
By "redefine signal" you mean the pin assignment on the board?

Thans again,

UCTRONICS, 2017-05-08:
@Lornioiz 
Yes.
Regards.

Lornioiz, 2017-05-08:
Some godd news!

I was able to make it work... more or less.
I compiled the sketch using the sketch "ArduCAM_ESP8266_OV2640_Capture.ino" under "Examples --> arduCAM -> ESP8266" and using a "NodeMCU 1.0" as board in Arduino ide.

However, I have image corruption in lower half of the screen. I have sneen that others have had the same problem and you usually suggest to try the latest drivers, but I already have them. I dowloaded the drivers and put the folder here "C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduCAM".
There is some other workaround for images corruption?

Lornioiz, 2017-05-08:
Ok, I was able to get rid of the artefacts lowering the: **static const size_t bufferSize = 2048;**
What are the problem lowering that value? can I have problems with higher resolutions?

Thanks!

UCTRONICS, 2017-05-09:
@Lornioiz 
Hi,

If you set the buffer value too low,it will influence your transfer speed and make the speed slow.
You can add some yield() in your code where transfers datas. 

let us know if you need more help.
Regards,
ArduCAM support team.

AlexandreN84, 2017-07-26:
Hi @UCTRONICS.

I updated my boards manager URLS, as you mentioned.
But I still keep the error:

> Can't find OV2640 module! pid: 254

What else can I try to do to resolve this error?

UCTRONICS, 2017-07-26:
@AlexandreN84 
Hi,
Which demo do you use?

AlexandreN84, 2017-07-26:
@UCTRONICS, thnaks for reply!
 
I'm currently using 
[https://gist.github.com/AlexandreN84/1d0e80746fb216129478eabf186b60ab](this scheme)

AlexandreN84, 2017-07-26:
@UCTRONICS, thnaks for reply!
 
I'm currently using 
[sssssssssss](https://gist.github.com/AlexandreN84/1d0e80746fb216129478eabf186b60ab)

AlexandreN84, 2017-07-27:
@UCTRONICS ,

I'm sorry, I was wrong with the codes.
The module in which I'm trying the [https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV2640_Capture/ArduCAM_ESP8266_OV2640_Capture.ino]("ArduCAM_ESP8266_OV2640_Capture")

The error it generates is as follows:
```
Can't find OV2640 module!

Connecting to mywifi
........WiFi connected

192.168.0.18
Server started
```

AlexandreN84, 2017-07-29:
@UCTRONICS,  
I found the solution.

The problem is [mentioned here](https://github.com/ArduCAM/Arduino/issues/92)

Thank you!