ArduCAM/Arduino #187: ESP8266 and OV5642 mini ARDUCHIP_TRIG, CAP_DONE_MASK issue

wdouglas1999, 2017-05-03:
Good day, 
I am using an ESP8266 and OV5642 mini. I am taking photos and storing them to the 8Mb FIFO.
Then i put the ESP8266 to sleep. The OV5642 remains powered.
The code runs perfectly the first time. Then when the esp8266 is reset the FIFO reads zero. Then i reset  the esp8266 again and everything works. So it works on every second reset.
Do you have any idea of what can be the issue? I suspect something to do the this line "while ( !myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))"  but i am not sure how to check it or reset the CAP_DONE_MASK. 

The code is below:  

void TakePics() {

myCAM.OV5642_set_JPEG_size(OV5642_640x480);delay(10);  ///set camera resolution
myCAM.write_reg(ARDUCHIP_FRAMES, 0x07);  ///set camera to take pictures until FIFO is full
myCAM.wrSensorReg16_8(0x4407, 0x08);   ///// set picture quality to medium. 02 is good 08 is low  
delay(10);  

  
Serial.println(F("Starting to take 100 photos")); 
myCAM.clear_fifo_flag();
delay(10);
myCAM.flush_fifo();
delay(10);
myCAM.clear_fifo_flag();

//Start capture
delay(100);

myCAM.start_capture();
///start_capture();
while ( !myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)){
#if defined (ESP8266)
 yield();
#endif
 } ; 

Serial.println(F("photos taken"));
}

UCTRONICS, 2017-05-04:
@wdouglas1999 
Hi,
Our camera don't support this mode you mentioned,
About capturing multiply pictures,please refer to this code we attached:

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use the enhanced functions
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It can  continue shooting and store it into the SD card  in JPEG format
// The demo sketch will do the following tasks
// 1. Set the camera to JPEG output mode.
// 2. Capture a JPEG photo and buffer the image to FIFO
// 3.Write the picture data to the SD card
// 5.close the file
//You can change the FRAMES_NUM count to change the number of the picture.
//IF the FRAMES_NUM is 0X00, take one photos
//IF the FRAMES_NUM is 0X01, take two photos
//IF the FRAMES_NUM is 0X02, take three photos
//IF the FRAMES_NUM is 0X03, take four photos
//IF the FRAMES_NUM is 0X04, take five photos
//IF the FRAMES_NUM is 0X05, take six photos
//IF the FRAMES_NUM is 0X06, take seven photos
//IF the FRAMES_NUM is 0X07, continue shooting until the FIFO is full
//You can see the picture in the SD card.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV5640_MINI_5MP_PLUS or OV5642_MINI_5MP_PLUS platform.
#if !(defined (OV5640_MINI_5MP_PLUS)||defined (OV5642_MINI_5MP_PLUS))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define   FRAMES_NUM    0x06
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
#define SD_CS 9
bool is_header = false;
int total_time = 0;
#if defined (OV5640_MINI_5MP_PLUS)
  ArduCAM myCAM(OV5640, CS);
#else
  ArduCAM myCAM(OV5642, CS);
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
Serial.begin(115200);
Serial.println(F("ArduCAM Start!"));
// set the CS as an output:
pinMode(CS, OUTPUT);
// initialize SPI:
SPI.begin();
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
[... 135 lines cut ...]
    myCAM.CS_HIGH();
    //Create a avi file
    k = k + 1;
    itoa(k, str, 10);
    strcat(str, ".jpg");
    //Open the new file
    outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
    if (! outFile)
    {
      Serial.println(F("File open failed"));
      while (1);
    }
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();   
    buf[i++] = temp_last;
    buf[i++] = temp;   
  }
}
  myCAM.CS_HIGH();
  return 1;
}

Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-05-04:
Hi, thank you for your reply. I do not understand what mode you are referring to. The issue is it works fine the first time and after resetting the esp8266 it does not. Something is not being reinitialized in the arducam chip.

closed 2017-06-01 by wdouglas1999