ArduCAM/Arduino #188: OV5642 mini Plus write multiple pictures to 8Mb FIFO

wdouglas1999, 2017-05-04:
Good day,
I am trying to write multiple pictures to the FIFO without success. I would like to be able to write about 80 pictures to the FIFO and then read it back. I am aware that i can write up to 6 pictures and then the entire buffer using FRAME_NUM but this is not what is want.
When i write to the FIFO the picture is overwritten and only the last one remains.
Looking at the .CPP and .H file i see these lines below relating to the FIFO write pointer. How ever in the  OV5642 mini application note the write pointer is bit 05 ?

#define ARDUCHIP_FIFO      		0x04  //FIFO and I2C control
#define FIFO_CLEAR_MASK    		0x01
#define FIFO_START_MASK    		0x02
**#define FIFO_RDPTR_RST_MASK     0x10
#define FIFO_WRPTR_RST_MASK     0x20**

This leads to a lot of confusion and it makes it very difficult to adopt the arducam for my application.
Where is the FIFO write pointer being reset and how can i avoid this from happening? 
Is it at all possible ,it yes , how?

Thank you kindly.

UCTRONICS, 2017-05-05:
@wdouglas1999 
Hi,

Yes, when the 0x04 register bit[4] is written to 1, it will reset FIFO write pointer.
When the bit[5] is written to 1,it will reset FIFO read pointer.
In our demo, we don't control the two bits,when you set the FRAME_NUM to 0x07,
It will capture until writting full 8MB. During this time ,you can not control the bit[4]
to reset the FIFO write point.
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-05-05:
Hi , thank you for the response.
Could you please tell me what command I would use to 
Firstly write '1' to the FIFO pointer and then how to clear it to zero.
Thank you

wdouglas1999, 2017-05-05:
Good Day,
By now i have spent all day trying to write multiple pictures to the FIFO ,which by now is starting to seem impossible. It seems that the write pointer is reset by the arduchip and cannot be changed by external mcu. Once the Arduchip has written to the FIFO it resets the Write pointer automatically.
Is this correct? Or is there any hope of controlling this externally? If not, it is really not a very flexible system for applications.
thank you.

UCTRONICS, 2017-05-07:
@wdouglas1999 
Do you use our ArduCAM_Mini_5MP_Plus_Multi_Capture2SD demo?
does it can work fine?
you can not set the FRAME_NUM more than 7.

Let us know if you need more help.
Regards,
ArduCAM support team.

wdouglas1999, 2017-05-07:
Hi,
Yes I use the demo and it works fine. The issue is I do not want to write to sd card. I would like to write from fifo to my server directly as a stream. So I want to write pictures to FIFO for 10 seconds then upload only the 10 seconds of pictures. I do not want to wait for the entire FIFO is full. It takes to long to fill.

Is it not possible to change a few lines of code in the firmware to allow for this? 
Maybe add a multiplier for FRAMENUMS to enable for example FRAMENUMS X 20 ?

UCTRONICS, 2017-05-08:
@wdouglas1999 

Hi, 
at present our camera can't do it.
If you set the FRAMENUMS to 7, 
you must wait capture done.

Let us know if you need more help.
Regards,
ArduCAM support team.