ArduCAM/Arduino #191: OV2640 - CAP_DONE_MASK is never set + a random question

thehult, 2017-05-11:
Hi!

I am working on a NodeJS project on a raspberry pi which uses the ArduCam Mini 2MP. Since I'm not that experienced with wrapping a c-library to Node, I decided to just "port" the code of the examples to js. I have followed the execution flow of all the examples for the mini, and written code that sends the exact same things to the mini. Problem is: I don't get past the corresponding line for:
`while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));`
 I have validated that the spi and i2c interfaces work as expected, so that is not the problem. I have also double checked that all the commands are correct (probably not since it is not working, but I can't find where).

The code looks as follows:
```
ArduCam.initJPEG = function() {
    ArduCam.writeSensorReg8(0xFF, 0x01);
    ArduCam.writeSensorReg8(0x12, 0x80);
    rpio.msleep(100);
    ArduCam.writeSensorReg8Arr(ArduCam.JPEG_INIT);
    ArduCam.writeSensorReg8Arr(ArduCam.YUV422);
    ArduCam.writeSensorReg8Arr(ArduCam.JPEG);
    ArduCam.writeSensorReg8(0xFF, 0x01);
    ArduCam.writeSensorReg8(0x15, 0x00);
    ArduCam.writeSensorReg8Arr(ArduCam.JPEG_320x240);
    rpio.msleep(1000);
};

ArduCam.takePicture = function() {
    ArduCam.writeReg(0x04, 0x01);
    ArduCam.writeReg(0x04, 0x01);
    ArduCam.writeReg(0x04, 0x02);
    var reg41 = ArduCam.readReg(0x41);
    while((reg41 & 0x08) == 0) {
        console.log("Waiting for picture");
        console.log("Reg 41: " + reg41 + ", anded: " + (reg41 & 0x08));
        rpio.msleep(2000);
        reg41 = ArduCam.readReg(0x41);
    }
    var len1 = ArduCam.readReg(0x42);
    var len2 = ArduCam.readReg(0x43);
    var len3 = ArduCam.readReg(0x44) & 0x7f;
    var length = ((len3 << 16) | (len2 << 8) | len1) & 0x07fffff;
    console.log("Picture length: " + length);
    if(length >= 0x5FFFF){
        console.log("Picture length too big.");
        return undefined;
    }
    if(length == 0){
        console.log("Picture length is 0.");
        return undefined;
    }
    rpio.write(ArduCam.chipSelect, rpio.LOW);
    rpio.msleep(100);

    var buf = Buffer.from([0x3C]);
    rpio.spiWrite(buf, buf.length);

    buf = Buffer.alloc(length);
    var respBuf = Buffer.alloc(buf.length);
    rpio.spiTransfer(buf, respBuf, buf.length);

    rpio.msleep(100);
    rpio.write(ArduCam.chipSelect, rpio.HIGH);
    return respBuf;
};

ArduCam.testSPI = function() {
    ArduCam.writeReg(0x00, 0x55);
    var test = ArduCam.readReg(0x00);
    return test == 0x55;
};

ArduCam.testI2C = function() {
    ArduCam.writeSensorReg8(0xFF, 0x01);
    var vid = ArduCam.readSensorReg8(0x0A);
    var pid = ArduCam.readSensorReg8(0x0B);
    return !((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 )));
};
```
And the flow is something like:
```
if(ArduCam.testSPI() && ArduCam.testI2C()) {
    ArduCam.initJPEG();
    ArduCam.takePicture();
}
```

I did not include the read and write functions, since they are working and pretty self-explaining. The ArduCam.JPEG_INIT, ArduCam.YUV422, etc. are of course the exact same as the ones you have in your examples.

One clue to the problem may be that bit 0 on address 0x41 sometimes is set and sometimes unset, indicating that vsync is active.

Another question, that might have something to do with it, but probably not: Why is the methods
`void ArduCAM::flush_fifo(void)` and void `void ArduCAM::clear_fifo_flag(void )` doing the same thing? To me it seems like `flush_fifo` should write 1 to bit 4 and 5 of 0x04?

I hope we can find the problem, and thanks in advance for the help!
Regards
TheHult

wdouglas1999, 2017-05-11:
@TheHult,

Hi, im no expert,but i had a quick look at your code and i saw ,
ArduCam.takePicture = function() {
    ArduCam.writeReg(0x04, 0x01);
    ArduCam.writeReg(0x04, 0x01);
    ArduCam.writeReg(0x04, 0x02);
    var reg41 = ArduCam.readReg(0x41);
    while((reg41 & 0x08) == 0) {
        console.log("Waiting for picture");
        console.log("Reg 41: " + reg41 + ", anded: " + (reg41 & 0x08));
        rpio.msleep(2000);
        reg41 = ArduCam.readReg(0x41);
    }

Is this a typo or are you meant to repeat this line
 ArduCam.writeReg(0x04, 0x01);
 ArduCam.writeReg(0x04, 0x01);

Also, i am also confused about the void ArduCAM::flush_fifo(void) and void void ArduCAM::clear_fifo_flag(void ) 

Great question.

thehult, 2017-05-11:
The repeating lines are a result of flush_fifo and clear_fifo_flag both writing 0x01 to 0x04, since I didn't know if this was intentional or not. I have tried different values for the flush method, but it didn't work either

nickaxgit, 2024-06-26:
7 years later, and I have hit exactly the same problem porting to TinyGo .. shame there are no answers