ArduCAM/Arduino #194: Arducam Mini Shield ov2640 SDCard error on Arduino and Esp12E

ayoadejoe, 2017-05-14:
Hello, I successfully set up arducam mini with Arduino and took shots with arducam.exe. I also successfully used it with esp12e/esp8266 video streaming and capture via browser.
The issue is, I need to make it work with SDCard. I used different Chip select as expected, made sure both are not low at the same time, nevertheless it keeps returning interface error.
I used same mosi, miso, clk  but used different cs on both uno and 8266, same error.
How can I resolve this?

supprot, 2017-05-14:
@ayoadejoe 
Hi,
What's your  SD CS?
Regards,

ayoadejoe, 2017-05-14:
@supprot 
On Arduino uno, I used pin 7 for arducam and 4 for a sdcard 
On esp, I used gpio 16 (D1) for arducam and gpio 2 (D4) for sdcard

Both yielded the same error. However, whenever I remove the miso/mosi pin, the arducam runs but sdcard fails (of course).

ayoadejoe, 2017-05-15:
Hello, I successfully set up arducam mini with Arduino and took shots with arducam.exe. I also successfully used it with esp12e/esp8266 video streaming and capture via browser.
The issue is, I need to make it work with SDCard. I used different Chip select as expected, made sure both are not low at the same time, nevertheless it keeps returning interface error.
I used same mosi, miso, clk but used different cs on both uno and 8266, same error.
How can I resolve this?

supprot, 2017-05-16:
@ayoadejoe 
Hi,
You should better check your SD CS by the oscilloscope firstly.
Maybe it exist some hardware problems.

Let us know if you need more help.
Regards,
ArduCAM support team.

ayoadejoe, 2017-05-23:
Alright, here is the solution. I switched the SD Card to 3.3V supply. The 5V I was previously using was probably too high for it even though I had used the SD on 5V in another project. It works quite well now.

closed 2017-05-23 by ayoadejoe