ArduCAM/Arduino #196: ArduCAM with RaspPi

bs31, 2017-05-15:
Hi, 
I use 4 ArduCAMs 5MP PLUS with the ArduCAM 4 cameras adapter board connected to a RaspPi.
Running ov5462_capture or ov5462_4cams_capture produces SPI interface errors without any changes made to the system.
If I run ov5462_capture with just one camera connected, it works quite well.

Anyone experienced the same problem or found a solution for the unstable SPI connection?

Cheers,
Tobi

supprot, 2017-05-16:
@bs31 
Hi, 

You can try add another power for the adapter board.

Let us know if you need more help,
Regards,
ArduCAM support team.

bs31, 2017-05-24:
Hi @supprot ,

unfortunately adding another power supply doesn't fix the problem.

Some more ideas?
Cheers, 
Tobi