ArduCAM/Arduino #197: OV5642 ESP8266_nodemcu v3 SPI1 interface error!

efloresfontau, 2017-05-17:
Hello,

I am testing the ArduCAM_ESP8266_OV5642_Capture Arduino demo using the AP working mode, but it's not shown in my internet networks. 

Monitor Serie shows the following:

wdouglas1999, 2017-05-17:
Good day,
I am also using node mcu with arducam mini OV2640 V2 and i have exactly the same problem. I know the spi works since i've tested in the other spi cameras, so it would be interesting to know what the issue is here.

supprot, 2017-05-18:
@efloresfontau @wdouglas1999 
Hi,

Please check the SPI CS firstly.
In our demo we define the SPI CS 16.

Let us know if you need more more help.
Regards,
ArduCAM support teram.,

wdouglas1999, 2017-05-18:
Hi, thank you for the reply.
I can confirm I have SPI CS 16 connected as CS for the camera. 
Do you have any other ideas?

wdouglas1999, 2017-05-18:
I just had a look at the circuit diagram of of the nodemcu and there are some other components connected to GPIO 16. I will investigate further and update the ticket.

wdouglas1999, 2017-05-18:
I changed the GPIO_CS from 16 to GPIO 0 and the SPI error is still there. I'm out of ideas for now. Anybody else have any ideas?

efloresfontau, 2017-05-18:
This is the pinout for the ESP8266 nodeMCU v3 lolin is the following:
[image]

I checked my connections and code and the SPI is connected to GPIO16 as code says.

I am also out of ideas. There's an error writing or reading? Which register is ARDUCHIP_TEST1? 
<// initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHz

  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");>

wdouglas1999, 2017-05-18:
Hi, I am also out of ideas. Today I will try to use the esp8266 standalone with the arducam and sd card. I will update the ticket with my findings

UCTRONICS, 2017-05-18:
@wdouglas1999 

Hi,
We suggest you check your SPI bus separately.
Regards,

efloresfontau, 2017-05-18:
@wdouglas1999 @supprot 

Hello,

I solved the problem. I declared my hardware platform in memorysaver.h as OV5642_MINI_5MP and i am using OV5642_MINI_5MP_PLUS. 

Now I have another problem, the image is blocky. May it be some frequency settings or buffer size? Should I open another issue, and close this one?

Regards,

efloresfontau, 2017-05-18:
Hello,

With a buffer size of 2048 the blocks dissapeared, but the image is not focused. I will investigate how to tune the focus.

[image]

Regards,

wdouglas1999, 2017-05-18:
Hi, the camera might have been shipped unfocused. Try to manually turn the lense.

wdouglas1999, 2017-05-18:
Hi, so I hooked up a stand alone esp8266 with the ov2640 and a SD card. Still the same SPI error. Then I decided to terminate  SPI lines (mosi, miso ,sck) with a R100 ohm resistor to ground. Everything is working now.
I hope this help you too if you SPI becomes unstable if you are using more than one SPI device.

jahearnco, 2017-06-08:
Not sure if this helps, but I just got the same error(s) before I realized that I had improperly grounded the camera. Possible those resistors you added set ground reference that wasn't set already for the camera (i.e. it's potential might be floating?).

wdouglas1999, 2017-06-08:
Hi, thanks for replying. I think you are correct since I am using a breadboard setup with a bunch of wires. Wire length is also an issue. I will make PCB' s soon so I just added those resistors to gnd as a precaution.

jahearnco, 2017-06-08:
When using breadboards I've learned (after MUCH trial and error - mostly error) to connect different sections of the ground track with higher gauge jumper wires. 3 or 4 of them. Even if not needed, at least I know everything is at essentially the same ground.

steelgoose, 2017-11-23:
I just committed out the two blocks of code... It works, don't know if it will cause any problems.
```
/*
//Check if the ArduCAM SPI bus is OK
myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
temp = myCAM.read_reg(ARDUCHIP_TEST1);
if (temp != 0x55){
Serial.println(F("SPI1 interface Error!"));
while(1);
}

//Check if the ArduCAM SPI bus is OK
myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
temp = myCAM.read_reg(ARDUCHIP_TEST1);
if (temp != 0x55){
Serial.println(F("SPI1 interface Error!"));
while(1);
}
*/
```