ArduCAM/Arduino #199: OV2640 mini 2MP connection to DUE

cblx5, 2017-05-22:
Hi,

please confirm the connections of the OV2640 mini 2MP to a DUE.
I have seen some posts about it but not clearly explained and references
to made  ICSP when I think it was SPI header.

OV2640 -  DUE
==========
CS          -  10
MOS      -  MOSI  on SPI header
MISO     -  MISO on SPI header
SCK       -  SCK on SPI header
GND      - GND on SPI header
5V          -  5V   on SPI header !!!!!!!!   (this is the one of 3  not clear)
SDA        -  SDA1   (no pullup resistor  needed)
SCL         -  SCL1     (no pullup resistor  needed)

Thank you

cblx5, 2017-05-25:
Hi,
Got my  OV2640 mini 2MP  working with DUE   and using an example.   If it helps anyone...

1) CS was pin 7 in code - so changed from 7 to 10 ( was building my own wiring cable before delivery of camera)
2) the "5V" on camera module is plugged into 3.3V on DUE
3) All others as above.

closed 2017-05-25 by cblx5

i7-Ryzen, 2019-05-01:
Hi cblx5,

Is there a particular order for SPI? Thanks

i7-Ryzen, 2019-05-01:
Solved. Thanks anyway :-)