ArduCAM/Arduino #201: (ArduCAM-Mini OV5642 & RaspberryPi) Empty JPEG

pavik1994, 2017-05-25:
Hello.
I have ArduCAM-Mini OV5642 connect with RaspberryPi Compute Module by SPI and I2C. When I wrote
"sudo ./ov5642_capture -c test.jpg 320x240"
and i have:
"OV5642 detected.
Changed resolution to 320x240
Start Capture
CAM Capture Done
Reading FIFO and saving IMG
The length is 2584"
and in this moment program stop and my "test.jpg" is empty.
What can I do?

closed 2017-05-29 by pavik1994

hunterbryant, 2018-07-02:
Hey @pavik1994, I have been getting the same issue and was wondering if you solved it. I tried to debug the saving to SD card and noticed the code inside if statements to write to the sd were never called. Any help is appreciated

martinlang83, 2018-10-19:
@hunterbryant same issue here. could you fix it?

hunterbryant, 2018-11-19:
@martinlang83 👎

UCTRONICS, 2018-11-20:
@hunterbryant @martinlang83 
Arducam has updated new Spi camera library for Raspberry Pi.
Please refer to the new library from https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera