ArduCAM/Arduino #202: OV2640 functions example - upside down and left to  right

cblx5, 2017-05-25:
Hello

Could someone (or Lee) please explain?   When I run ArduCAM_Mini_2MP_OV2640_functions   with the windows host and set to  320_240 and jpeg,   I get quite a nice single shot photo.

However, if I tick "Set ToBMP"  (because I want  RGB565 output) ,   I still get a picture  but it is  upside down and left to right?

Its not a big  problem as I can write code to reverse it, but would lie to understand why, before delving deep into the code.

Thanks

dilpreetdan, 2017-06-01:
Hi,
Are you able to get a valid BMP image from the host app? In my case, when I inspect the BMP image, its just a JPG stored with BMP extension.
Can you inspect your BMP images in hex encoding and validate?

Thanks!

cblx5, 2017-06-02:
Hi,

I used the host app only to verify things were working   visually as I always  intended to display to a TFT.
So I did not investigate the host app too much

As the BMP image was clean although wrong orientation,  I proceeded to develop my code (and also got no response here).

However when I started to read the BMP image from the FIFO (buffer)  and display it on my TFT,  I found that the image was only left to right and not upside down.  

So pixel (0,0)  was stored in the FIFO at  (319,0)  as I read out the pixels.
and  pixel (0,1) was at (319,1) and so on.....

So instead of displaying to the TFT at (x , y )  it is displayed  at   ( 319-x ,  y ).

So the image on the TFT was exactly how I expected it....   not upside down not left to right.

So I just remember to put (319-x)  where I expected to only have x !!

Back to the host app  ......  at   320x240....

1) a jpeg image is about   17,000 bytes

2) a BMP image is about     153,600 bytes   (320x240x2)   plus the header data that the Host app needs of about 160 bytes   and some terminating (I  think about 2)  bytes.

so if the host app byte counter is   incrementing about 154,000 bytes each image,  you are getting a full BMP at 320x240 pixels.

Hope all that made sense.