ArduCAM/Arduino #205: OV2640 2MP Mini: Bitmap is not working

dilpreetdan, 2017-05-31:
Hi,
I am using OV2640 Mini camera with Arduino Uno R3. I have successfully made the connections and verified using the Arducam Host app that it works.
Now, I am using PySerial instead to send commands and read the values written back by the device (device has Arducam_Mini_Video_Streaming.ino). When setting the JPG mode and saving the image(using 0x00 and 0x10), I can see correct images. However, on setting the BMP format (using case 0x31 and then 0x30 to capture BMP), I get an incorrect distorted image. When I open the image as a HEX file, I can verify the header is correct and the file terminates with 0xBB and 0xCC.

On windows, using the Arducam Host app, I can save JPG and BMP images of different resolutions. However, upon investigating the bmp images, I find that they are jpeg images (case 0x10) saved with bmp extension.

So essentially, the BMP images with correct header are not being produced either by the ArducamHost app and neither by using the code.

Attaching the hex file which was generated. If you save this same file as a bmp format, you can see the distorted image.
[BMP_data.txt](https://github.com/ArduCAM/Arduino/files/1041974/BMP_data.txt)

Also attaching the hex file for a JPG image as reference.
[JPG_data.txt](https://github.com/ArduCAM/Arduino/files/1041967/JPG_data.txt)

Can you please help and suggest how to actually produce a BMP image (using Arducam_Mini_Video_Streaming)?

supprot, 2017-06-01:
@dilpreetdan 
Hi,
Please refer to the picture below:
[image]

dilpreetdan, 2017-06-01:
Hi,
Thanks for your reply. 

As I mentioned earlier, I took both JPG and BMP (using the ArducamHost) and upon checking the hex files of these images, you can verify that the BMP image does not correspond to BMP but rather JPG. It's header is not a BMP header.

Also, by sending 0x31 and 0x30 in the Arducam_Mini_Video_Streaming.ino, it doesn't work properly. 

I am aware of the working of Arducam Host. Please address the question as to how to make a bitmap image with correct BMP header.

supprot, 2017-06-03:
@dilpreetdan 
Hi,
Do you use this code?
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino
 
Don't worry and we will test it nest monday and give you a reply.

Let us know if you need more help.
Regards,
ArduCAM support team.

dilpreetdan, 2017-06-10:
Hello,
I'm awaiting your response. It would be great if you can resolve this as soon as possible. I have some work dependent on this.

Thanks!

17ec084, 2020-08-01:
@dilpreetdan 
Hello, I saw your BMP_data.txt and found that biCompression is not jpeg(04 00 00 00) but "BI_BITFIELDS"(03 00 00 00).