ArduCAM/Arduino #206: OV2640 2MP Mini: Take RAW Grayscale image

dilpreetdan, 2017-06-01:
Hello,
How can we take a grayscale RAW image from the OV2640 2MP Mini camera? I need a 160*120 image with 8 bits per pixel data.

Can you please provide some code for the same.

Thanks!

dilpreetdan, 2017-06-10:
Hello,
Any help in this would be really appreciated!

UCTRONICS, 2017-06-10:
@dilpreetdan 
Hi,
Very sorry for we do not provide source code of Host V2.

Let us know if you need more help.
Regards,
ArduCAM support team.

dilpreetdan, 2017-06-10:
Thanks for responding.

Firstly, I think this is another thread to reply to.
Secondly, I see that the source code of Version 1 was available.
Thirdly, using the source code of version 1, I could verify that the bitmap image is not getting produced. (https://github.com/ArduCAM/Arduino/issues/205)
Please rectify such problems so that a correct BMP image is produced.

Thanks!

UCTRONICS, 2017-06-10:
@dilpreetdan 
Hi,
We have tested our host_V2 and Video_Streaming code and they can work fine for getting BMP image .

Pleaase download our Host_V2 from https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows
Please download this code:
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming

Let us know if you need more help.
Regards,
ArduCAM support team.

dilpreetdan, 2017-06-10:
[image]

This should help you identify the problem. Please can you verify in your BMP image if the header is correct.

Thanks!

dilpreetdan, 2017-06-10:
Surprisingly, with Arducm Host v1.exe this is not the issue. I could verify that from the source code.
[image]

I think in the python script where I receive the data, I should be counting length <153664 and it should help with the BMP.

UCTRONICS, 2017-06-10:
@dilpreetdan 
Hi,
Thank you for your advice and we will optimize our host V2 as soon as quickly.
Regards.

dilpreetdan, 2017-06-19:
Hello,
Please provide some feedback on grayscale image. How do I take a grayscale image (like .PGM format).

Thanks!

nikita-kotsehub, 2021-07-26:
Any updates on the RAW grayscale image? @dilpreetdan @UCTRONICS