ArduCAM/Arduino #208: Arduino IED Not Supporting ArduCAM library

rosegfuzzy, 2017-06-03:
My Arduino IED will not recognize the ArduCAM library when I upload the zip file to add it to my libraries. Can anyone help me or provide me with a ArduCAM library that will be supported by the Arduino IED?

supprot, 2017-06-04:
@rosegfuzzy 
Hi,

You should put the ArduCAM folder instead of Arduino-master to your libraries.

Let us know if you need more help.
Regards,
ArduCAM support team.

rosegfuzzy, 2017-06-04:
I have tried that and I get the error that it does not contain a library. I had to add the unzipped folder of ArduCam for it to work.

rosegfuzzy, 2017-06-04:
Now the example code will not upload to my Arduino because they are tons of errors which I try to fix but causes more errors. Can you please send me a better version of the example code?

francescorinaldi, 2017-06-06:
@rosegfuzzy 

Follow these steps:
1) Unzip the Arduino-master.zip file
2) Move the ArduCAM folder into your ../Arduino/libraries
3) Run your Arduino application

Examples should work for you now.

rosegfuzzy, 2017-06-06:
Thank you for your advice. I already did that. My problem now is that the example sketch ArduCAM_Mini_5MP_Plus_Multi_Capture2SD will not compile due to the number of errors in the code. I have spent hours trying to fix it. It makes me fix errors then undo the fixing. Can you help me make the code error free or let me see a better version of the code?
Thanks,