ArduCAM/Arduino #209: ArduCAM with Arduino / Genuino 101

francescorinaldi, 2017-06-06:
Hi,
I'd like to use ArduCAM 5MP [OV5642] on a Arduino 101. Is it supported? Are the compatible?

If yes, where can I find a connection scheme or explanation about how connect them?

Thank you in advance!

francescorinaldi, 2017-06-06:
I tried with this configuration:
ArduCAM OV5642 -------------- Arduino / Genuino 101
CS -------------------------------------------- PIN 7
MOSI ----------------------------------------- PIN 11
MISO ----------------------------------------- PIN 12
SCK ------------------------------------------ PIN 13 
GND ------------------------------------------ GND
VCC ------------------------------------------ 5V
SDA ------------------------------------------ SDA
SCL ------------------------------------------ SCL

closed 2017-06-06 by francescorinaldi

supprot, 2017-06-06:
@francescorinaldi 
Hi,
Yes, The ArduCAM 5MP  support Arduino 101 board. 
We suggest you use our adapter board.

Let us know if you need more help.
Regards,
ArduCAM suppor team.