ArduCAM/Arduino #210: macOS Host Application

francescorinaldi, 2017-06-06:
The macOS Host Application link is missing. Can you please add it, to the latest version?

Thank you in advance.

supprot, 2017-06-06:
@francescorinaldi 
Hi,
www.arducam.com/downloads/app/ArduCAM_Host_V2.0_Mac.app.zip

If you have any problems,please refer http://www.arducam.com/knowledge-base/arducam-typical-problems-arduino-uno/
to find answer.

Let us know if you need more help.
Regards,
ArduCAM support team.

ProjectileObjects, 2018-08-10:
It doesn't work. MacOSX 10.12.6

bakerbrown, 2019-07-05:
Just download and its is working with Mojave 10.14.5. However it only captures JPG images and I need BMP Images that are not working correctly on the Windows version of the same software!

UCTRONICS, 2019-07-08:
@bakerbrown 
If you want to get BMP images, you can use our Window version here 
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows

hgarcia101685, 2021-03-14:
The MacOs one does not open properly on version Catalina 10.5.7.  Can you guys please post an update that will work on this version?  I don't mind if it's just JPGs.

closed 2022-10-29 by francescorinaldi