ArduCAM/Arduino #211: Sending/Uploading Arducam Images/Videos to my Website

ayoadejoe, 2017-06-06:
I have succeeded in posting String/Integer data to my website. I could use PHP to save those data to my database. The challenge I am facing now is how to send Images/Video files. 

I need help in writing a code PHP/HTML to act as Server on my site to accept the connection, read the file sent from the esp8266 and save it to data base or just my gallery.

Then I need help with a esp 8266 client code to write the file to the online server. I am thinking this should be fairly easy for a person good with html/php.

Kindly help with both server and client code. Please remember, I need esp to function on the client side.

For now, this is what I have done: 

`#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

const char* MY_SSID = "myssid";
const char* MY_PWD =  "mypassword";
const char* server = "http://mysite.com";
HTTPClient http;

void setup() {
  Serial.begin(115200);                 //Serial connection
  WiFi.begin(MY_SSID, MY_PWD);   //WiFi connection
 
  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion
    delay(500);
    Serial.print(".");
  }
}

void loop() {
 
 if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status
 
   HTTPClient http;    //Declare object of class HTTPClient
 
   http.begin(server);      //Specify request destination

   String HPost = getHttpPost();
   Serial.print("This is post going out:");
   Serial.println(HPost);
   http.addHeader("Content-Type", "text/plain");  //Specify content-type header
 
   int httpCode = http.POST(HPost);   //Send this request

   Serial.println(httpCode);   //Print HTTP return code

if(httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.println("code received");

      // file found at server
      if(httpCode == HTTP_CODE_OK) {
        String payload = http.getString();
        Serial.println(payload);    //Print request response payload
      }else{
          Serial.println("No payload");  
      }
    } else {
       Serial.print("Issues connecting to: ");
       Serial.println(HPost);
    }
 
   http.end();  //Close connection
 
 }else{
    Serial.println("Error in WiFi connection");   
 }
 
  delay(30000);  //Send a request every 30 seconds
 
}

int w=1;
String getHttpPost(){
    // Wait a few seconds between measurements.
  delay(2000);
 
String data = "B=11&C=22&D=33&E=44&F=55";
          return data;
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}`

My php code on my website that deals with my esp post data is based on this site: 
[https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&cad=rja&uact=8&ved=0ahUKEwjJ5LynsqnUAhVNkRQKHdQ6ATwQFggsMAE&url=https%3A%2F%2Fwww.hackster.io%2Fdetox%2Fsend-esp8266-data-to-your-webpage-no-at-commands-7ebfec&usg=AFQjCNGxQFlpdWocxepouhD0t4W10ctUUA&sig2=uGj5c0L0ZN9KwK6PY4dmkA](url)

bilbaof107, 2019-04-07:
Hey, were you able to figure this out? I also need help with this!

UCTRONICS, 2019-04-07:
@ayoadejoe 
Your project sounds great. At present, we can help you ensure the camera drivers normally and you can get the image data to the memory. What's more, we have a demo. The ESP8266 act as a web server and you can use the browser which acts as a client to get the image. You can get it from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/blob/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_UNO_Capture/ArduCAM_ESP8266_UNO_Capture.ino

Can you describe your detail ideal and maybe our engineer can help you?