ArduCAM/Arduino #212: OV5642 problem with SPI

pavik1994, 2017-06-08:
Hi,
i have problem with my ArduCAM. I connect with RaspberryPi and level voltage in SPI is higher than 3 V. When camera is connect with RaspberryPi in SPI pin is 3V3 but when I started program this level up to 4 V. What can I do with this? 
This problem is with power camera woth 5V and 3V3. 
I tried use camera in few RaspberryPi so this isn't problem with RaspberryPi.
Thank you for help :)

pavik1994, 2017-06-12:
@ArduCAM

supprot, 2017-06-15:
@pavik1994 
Hi,
Sorry for my later reply.
Please connect the camera to 3.3V instead of 5V .

Let us know if you need more help.
Regards,
ArduCAM support team.

pavik1994, 2017-06-15:
Hi,
I connect the camera to 3.3V but still in SPI is 4V or more. What can I do
more?

Regards

wdouglas1999, 2017-06-16:
Hi, try adding 100 ohm resistors between the SPI lines and ground.