ArduCAM/Arduino #214: Source code for ArduCAM_Host_V2.exe

dilpreetdan, 2017-06-10:
Is the source code for ArduCAM_Host_V2.exe available somewhere? Also, please update the links and content on the website here (http://www.arducam.com/arducam-host-application-demonstration/)

jason66, 2017-09-20:
I would like source code for ArduCAM_Host_V2.exe  as well.

orion434, 2021-02-01:
Please, share the source code or links to existing C(++) files to be compiled.
The app is bugged, so at least I could fix it myself!

UCTRONICS, 2021-02-02:
Please contact my email **support@arducam.com** and I will send you source code. Thanks for your debug for us.

orion434, 2021-02-25:
Dear @UCTRONICS , I sent an email three weeks ago but I never got any reply. Can you please check if it did arrive correctly?