ArduCAM/Arduino #215: Arducam Shield V2 tutorial

dilpreetdan, 2017-06-10:


supprot, 2017-06-16:
@dilpreetdan @nilslidbaum 
Hi ,
Sorry for reply you so late.

Please refer to this link:
http://www.arducam.com/downloads/shields/ArduCAM_Shield_V2_Hardware_Application_Note.pdf
http://www.arducam.com/downloads/shields/ArduCAM_Software_Application_Note.pdf

Let us know if you need more help.
Regards,
ArduCAM support team.

dilpreetdan, 2017-06-16:
Thank you for your response.

I hadn't copied the UTFT4ArduCAM_SPI library and hence was facing problems. Now I have both Arducam and UTFT4ArduCAM_SPI libraries in the Arduino/libraries/ folder and the example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Digital_Camera works!