ArduCAM/Arduino #217: Panoramic Camera Issue

headofled, 2017-06-13:
So I'm trying to build the panoramic camera found here:
http://www.arducam.com/arduino-panorama-photography-arducam/
I bought 4 OV5642_5MP_Plus cameras, and an Arducam Multi Camera Adaptor Board, both for an Arduino UNO (I plan on upgrading to a MEGA in the near future, preferably after I get it to work on an UNO. 
I downloaded and installed the Arducam libraries needed for the project, and altered "memory.h" in Microsoft Visual Studio so that the code would be compatible with cameras. However when I try to run the example codes for the Arducam Mini_5MP_Plus cameras, I get this error:

Arduino: 1.6.10 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Herlihyi\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10610 -build-path "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Mini_5MP_Plus\ArduCAM_Mini_5MP_OV5642_Plus_Functions\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Herlihyi\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10610 -build-path "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Mini_5MP_Plus\ArduCAM_Mini_5MP_OV5642_Plus_Functions\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino"
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp\sketch\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp\sketch\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp\sketch\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\Herlihyi\AppData\Local\Temp\build97f5b9a046f0ebea998a0e26182e8be5.tmp\sketch\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\Wire.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility\twi.c" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\libraries\ArduCAM\ArduCAM.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src" "-IC:\Program Files (x86)\Arduino\libraries\ArduCAM" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Program Files (x86)\Arduino\hardware\arduino\a [... cut ...]
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library ArduCAM in folder: C:\Program Files (x86)\Arduino\libraries\ArduCAM (legacy)
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
exit status 1
'OV5642_CHIPID_HIGH' was not declared in this scope

What should I do? Is there something in memory.h that I forgot to comment/uncomment? I could use all the help I can get.

supprot, 2017-06-16:
@headofled 
Hi, sorry for reply you so late.
If you are using 5MP_plus,you should enable OV5640_MINI_5MP_PLUS and disable others.

Let us know if you nee more help.
Regards,
ArduCAM support team.

headofled, 2017-06-18:
Better late than never.

I did that, but now I'm getting this error:

Arduino: 1.6.10 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Mini_5MP_Plus\ArduCAM_Mini_5MP_OV5642_Plus_Functions\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void setup()':

ArduCAM_Mini_5MP_OV5642_Plus_Functions:66: error: 'OV5642_CHIPID_HIGH' was not declared in this scope

     myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);

                           ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:67: error: 'OV5642_CHIPID_LOW' was not declared in this scope

     myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);

                           ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:80: error: 'VSYNC_LEVEL_MASK' was not declared in this scope

   myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH

                                 ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:84: error: 'ARDUCHIP_FRAMES' was not declared in this scope

 myCAM.write_reg(ARDUCHIP_FRAMES,0x00);

                 ^

C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Mini_5MP_Plus\ArduCAM_Mini_5MP_OV5642_Plus_Functions\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void loop()':

ArduCAM_Mini_5MP_OV5642_Plus_Functions:160: error: 'VSYNC_LEVEL_MASK' was not declared in this scope

     myCAM.clear_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);

                                   ^

exit status 1
'OV5642_CHIPID_HIGH' was not declared in this scope

headofled, 2017-06-18:
I did that, but it's still giving me 'OV5642_CHIPID_HIGH' not declared in scope

supprot, 2017-06-18:
@headofled 
Hi Ian,
Please attach us some pictures for your problems and we 
will try our best to help you solve your problems.
Which demo do you use? 

Let us know if you need more help.
Regards,
ArduCAM support team.

headofled, 2017-06-18:
I figured out the 'OV5642_CHIPID_HIGH' problem by adding the line:
#include "ov5642_regs.h"

Now I'm getting a error saying:
Arduino: 1.6.10 (Windows 8.1), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Herlihyi\AppData\Local\Temp\arduino_modified_sketch_200703\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void setup()':

ArduCAM_Mini_5MP_OV5642_Plus_Functions:81: error: 'VSYNC_LEVEL_MASK' was not declared in this scope

   myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH

                                 ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:85: error: 'ARDUCHIP_FRAMES' was not declared in this scope

 myCAM.write_reg(ARDUCHIP_FRAMES,0x00);

                 ^

C:\Users\Herlihyi\AppData\Local\Temp\arduino_modified_sketch_200703\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void loop()':

ArduCAM_Mini_5MP_OV5642_Plus_Functions:161: error: 'VSYNC_LEVEL_MASK' was not declared in this scope

     myCAM.clear_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);

                                   ^

exit status 1
'VSYNC_LEVEL_MASK' was not declared in this scope

Too answer your question, I'm using the ArduCam_Mini_5MP_OV5642_PLUS_FUNCTIONS demo.
I'm just trying to get 1 camera to work before getting 4 of them to work simultaneously.

Here are some pictures:
[image]
[image]

bs31, 2017-07-05:
Uncomment the `#define ARDUCAM_SHIELD_REVC `unless you are using this shield http://www.arducam.com/tag/arducam-rev-c/