ArduCAM/Arduino #219: Capture BMP image of different resolutions

dilpreetdan, 2017-06-15:
Hello,
From https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ov2640_regs.h I could verify that there are different resolutions available for taking a JPG image; however for a BMP image, only one: OV2640_QVGA i.e. 320*240.

Can you please provide similar register address-data configurations for taking QQVGA i.e. 160*120 or QQQVGA i.e. 80*60 BMP images?

Thanks,
Dilpreet

dilpreetdan, 2017-06-22:
Please can you provide some update on this?

dilpreetdan, 2017-07-18:
Hello,
Can you answer whether it is even doable for our camera or not. Its been a while and there is no response from your end.

Thanks!

supprot, 2017-07-18:
@dilpreetdan 
Hi,
Very sorry for reply you so late,
Yes,  it is  doable for our camera but at present we don't  have demos
 about configuring it in other resolutions except 320x240.

Let us know if you need more help.
Regards,
ArduCAM support team.