ArduCAM/Arduino #221: Rewind operation

dilpreetdan, 2017-06-16:
Hello,
I am unsure how to use the rewind read operation as provided here: http://www.arducam.com/tag/arducam-mini/

According to the software application note pdf Page 10 (http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf) it says: 

> By sending the command code 0x84 and write ‘1’ to bit[5] in the data phase, it will reset the
> memory read pointer to ZERO. 

However, from the hardware application note pdf (http://www.arducam.com/downloads/shields/ArduCAM_Mini_2MP_Camera_Shield_Hardware_Application_Note.pdf), it says:

> 0x04           |             RW             |          Bit[4]: write ‘1’ to reset FIFO write pointer 
>                                                                Bit[5]: write ‘1’ to reset FIFO read pointer 

In the Arducam.h file (https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.h), there is an inconsistency in the mask for read and write pointers (Lines 527-528)
>  #define FIFO_RDPTR_RST_MASK     0x10      // This is Bit[4] = 1
>  #define FIFO_WRPTR_RST_MASK     0x20     // This is Bit[5] = 1

Please check about this inconsistency and whether its 0x84 or 0x04. Also, can you please provide a code snippet for rewind read operation.

Thanks in advance!

supprot, 2017-06-18:
@dilpreetdan 
Hi,
The define in the library is right,
You can refer to this code
uint8_t read_fifo_again(ArduCAM myCAM)
{  
  uint8_t temp, temp_last;
  uint32_t length = 0;
  temp = myCAM.read_reg(ARDUCHIP_FIFO);
  myCAM.write_reg(ARDUCHIP_FIFO, temp|FIFO_RDPTR_RST_MASK);  //FIFO_RDPTR_RST_MASK
  length = myCAM.read_fifo_length();
  Serial.println(length, DEC);
  if (length >= FIFO_SIZE) //8M
  {
    Serial.println("Over size.");
    return 0;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println("Size is 0.");
    return 0;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  SPI.transfer(0x00);//First byte is 0xC0 ,not 0xff
  length--;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
   if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(12);
  }
  myCAM.CS_HIGH();
  is_header = false;
}

Let us know if you need more help.
Regards,
ArduCAM support team.

dilpreetdan, 2017-06-28:
Is there a way to manually random access a pixel in the FIFO buffer of the camera, instead of resetting the read pointer to start of the fifo? I would like to skip to a particular pixel in the buffer and do some processing on it without having to bring the entire image onto arduino pixel by pixel.

Thanks!

dilpreetdan, 2017-07-19:
Hello,
Can you please provide an answer to the above query?

> Is there a way to manually random access a pixel in the FIFO buffer of the camera, instead of resetting the read pointer to start of the fifo? I would like to skip to a particular pixel in the buffer and do some processing on it without having to bring the entire image onto arduino pixel by pixel.
>

ArduCAM, 2017-07-19:
According to FIFO structure, you can not random access the particular pixel.
So you have to read from the beginning of FIFO and drop unwanted pixel values.