ArduCAM/Arduino #222: Mini 2MP V2 with Arduino Mega

jessicayin, 2017-06-16:
Hi, 

I've connected the mini 2mp v2 to an Arduino Mga 2560 per your wiring recommendations:

* MOSI/MISO/SCK to ISCP pins 
* CS to pin 10 (PWM)
* SCL/SDA to SCL/SDA (20/21 COMMUNICATION)
* VCC to 5V
* GND to GND

And I followed this guide: 
http://www.arducam.com/how-to-connect-arducam-shield-to-mega-2560/
to make changes to the code (SD library, input/output pins in the Arduino sketch).

And I tried running:

* ArduCAM_Mini_2MP_OV2640_functions
* ArduCAM_Mini_Video_Streaming
* ArduCAM_Mini_LowPowerMode

However, I just get a black screen on the window of the host app (tried all versions, including those in th Windows 2.0 folder). It says "COM IS OPEN" and "320x240 is OK", as well as "Continuous is OK!" and "ArduCam continuous", but there are no images. How can I get the image to show up on the host app?

Thanks!

dilpreetdan, 2017-06-17:
HI,
V2 means shield version 2?

Can you try connecting VCC and GND to the pins in ISCP (there are 6 pins in ISCP with GND and VCC also).

supprot, 2017-06-18:
@jessicayin 
Hi,
 Can you find "SPI is OK" and "OV2640 is detected"  in the Host V2?
Regards,

jessicayin, 2017-06-19:
@dilpreetdan 
Thanks for your response. I tried connecting the VCC and GND to pins in the ISCP, but it didn't make a difference.

@supprot 
Good point, neither of those phrases appear in the Host V2. I tried changing the baud rate to 921600 on the Host V2, and now I can get "ArduCAM start!" (which is new!) but then a bunch of "SPI interface Error". The following code is the addition I made to the Arduino sketch (following the Arduino Mega guide I linked in my original post):

`void setup() {
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);`

jessicayin, 2017-06-19:
**edit: forgot to include
`pinMode(53, OUTPUT);`
as part of the code I added. I also tried running this sketch with the wires connected to the digital pins (51, 52, 53).

supprot, 2017-06-22:
@jessicayin 
Hi,

Please refer to this link Q9
http://www.arducam.com/knowledge-base/arducam-typical-problems-arduino-uno/
Let us know if you need more help.
Regards,
ArduCAM support team.