ArduCAM/Arduino #223: Displaying BMP on screen

giv86, 2017-06-18:
Greeting how can I display some BMP file from SD card on display?

supprot, 2017-06-18:
@giv86 
Hi,
At present , our Host_V2 exist some bugs about display BMP picture.
You can use our Host V1 to display firstly.
If you want to display the picture from your SD card, you can read it and display it then,
In detail , please refer to this code which is display jpeg picture from SD card.
image = SD.open("pic.jpg"); 
imageSize = image.size();
Serial.println(F("ACK CMD Opened image"));
Serial.println(F("ACK IMG"));
while(image.available()) {
if(imageSize--){
Serial.write(image.read());
delayMicroseconds(20);
}
else
image.close();
}
image.close(); 
Serial.println(F("ACK CMD picture send OK"));

Let us know if you need more help.
Regards,
ArduCAM support team.

giv86, 2017-06-18:
oh sorry to confused you, i was totaly meaning about LCD screen attached to your v2 shield
what i'm trying to acomplish is graphical menu with bacxkgrounds readed from SD card.

giv86, 2017-06-19:
@supprot please answer :)

supprot, 2017-06-22:
@giv86 
Hi,
Sorry for reply you so late ,
we don't have some example but you can refer to this code to get some help.
https://github.com/Smoke-And-Wires/TFT-Shield-Example-Code

Let us know if you need more help.
Regards,
ArduCAM support team.