ArduCAM/Arduino #224: OV5642 5MP Plus Not Detected

isaacviel, 2017-06-18:
Hello, 

I've just received my `Arducam Mini Module Camera Shield 5MP Plus OV5642 Camera Module` and having an issue with it not being detected. 

This is my full set up:
- Arduino UNO R3
- Arducam Mini Module Camera Shield 5MP Plus OV5642 Camera Module
- Arducam Multi Cameras Adapter Board for Arducam Mini Board

I currently have the single OV5642 plugged into the Arduino Mini 4 pins on the mini board and am running the following sketch: `ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino`

```
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can test OV5642 functions.
//

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV5642_MINI_5MP_Plus  platform.
#if !(defined OV5642_MINI_5MP_PLUS)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
 ArduCAM myCAM( OV5642, CS );
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
  // put your setup code here, to run once:
  uint8_t vid, pid;
  uint8_t temp;
  #if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
  
  #else
  Wire.begin();
  Serial.begin(921600);
  #endif
  
  Serial.println(F("ACK CMD ArduCAM Start!"));
  // set the CS as an output:
  pinMode(CS, OUTPUT);
  // initialize SPI:
  SPI.begin();
  
  while(1) {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
[... 848 lines cut ...]
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
	//#define OV2640_CAM
	//#define OV3640_CAM
	//#define OV5642_CAM
	//#define OV5640_CAM
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM
	//#define MT9M001_CAM
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif

#endif	//_MEMORYSAVER_
```

I had this working a few days ago and was able to take pictures and now it's dead. Any ideas?

[image]

[image]

isaacviel, 2017-06-20:
@ArduCAM any ideas on this?

supprot, 2017-06-22:
@isaacviel 
Hi,
Please check your camera firstly,Can your camera work fine at other platform?
Let us know if you need more help.
Regrads,
ArduCAM support team.

isaacviel, 2017-06-22:
@supprot Thanks for the reply. Putting the OV5642 module in `Arducam Mini 1`, `ArduCam Mini 2` or `ArduCam Mini 3` results in the following error: `ACK CMD SPI interface OK.` It appears that only `ArduCam Mini 4` is functional.

isaacviel, 2017-06-22:
@supprot @ArduCAM I just tried running the module directly from the Arduino Uno R3, changing
```
const int CS = 7; 
```
to
```
const int CS = 10;
```

with the same issue: 

`COM is open!

ACK CMD ArduCAM Start!
ACK CMD SPI interface OK.`

after which nothing else happens and nothing is either detected or not detected. I'm really stumped here and starting to thing the module is just simply dead. 

[image]

supprot, 2017-06-22:
@isaacviel 
Hi,
Which demo do you use? we will help you test it again.
Regards,
ArduCAM support team.

isaacviel, 2017-06-22:
@supprot

I've tried every sketch in the example library with the same results: nothing happening after the SPI interface check.

UCTRONICS, 2017-06-24:
@isaacviel 
 Hi,
Sorry for reply you so late. Maybe your camera exist some defects with I2c interface.
Have you tried it on other platform?

Let us know if you need more help.
Regards,
ArduCAM support team.

isaacviel, 2017-06-26:
@UCTRONICS @supprot 

I have tried on 2 other Arduino Uno R3s and get the same issue each time. I believe this new OV5642 is dead. Is there a way I can test it to be sure?

UCTRONICS, 2017-06-28:
@isaacviel 
Hi,
You can print the PID and VID calue after  
   myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
if you are using 2MP, the vid is 0x26 and the pid is 0x41 or 0x42.

Let us know if you need more help.
Regards,
ArduCAM support team.

Dickoabc123, 2023-07-09:
Hey there, did you make any progress with this, I’m have the same issue unfortunately.