ArduCAM/Arduino #225: Displaying OV2640 images on computer + 2nd generation ArduCAM Shields

tonehm, 2017-06-24:
Hello, 
I have connected a OV2640 camera with an Arduino UNO and a V2 Shield and got the system working with the codes provided here. However, I have only managed to take a picture that will save at a microSD card. I would like to send images (no need for video) to my computer. I have tried to find a tutorial for this. I see there are files under the example "host", but I can't find an explanation of how to use these. I saw in another post that you answered that there is an example code/program of this, but I can't find it. 

Additionally, I have the second generation of ArduCAM OV2640 camera with embedded shield + an SD expander for this. I don't have much experience with circuits I can't figure out how to wire it up, especially not with the SD-card expander. However, if there is a way to just send the images directely to my computer (as in the first paragraph) that would be preferrable and I wouldn't need the SD expander at all.

Would really appriciate any help on this!

UCTRONICS, 2017-06-24:
@tonehm 
Hi,
If you wnat to send image to your computer,you should use our Host_V2 from tis link:
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows
What's more, you should use this demo 
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino

Let us know if you need more help.
Regards,
ArduCAM support team.

tonehm, 2017-06-26:
The second link you posted is in the folder of the "mini" cameras. Will it still work for the "not-mini" cameras? Like OV2640, MT9D111, for example.

Another question; I have two cameras with fewer pins than the V2 Shield (OV7076 and OV9655). I did not get it to work with the shield and code package as posted here. Is it not supposed to work, and is it potentially another alternative shield for these cameras?