ArduCAM/Arduino #226: While files do I download?

headofled, 2017-06-27:
So I've been posting on several forums about issues I've been having with OV5642_5MP_Plus cameras with Arduino, but haven't had much luck. I've decided to just uninstall the libraries for Arducam and reinstall them, as I probably didn't install it properly. My question is, what exactly do I install? In case you're wondering, I'm using an Arduino UNO (will upgrade to a Mega), with an Arducam 4 camera adapter board, and 4 OV5642_5MP_Plus cameras. If anyone can point me in the right direction, that would be great.

UCTRONICS, 2017-06-28:
@headofled 
Hi, 
You should download this library and put the ArduCAM library to your libraries folder.

https://github.com/ArduCAM/Arduino.git

Regards,
ArduCAM support team.

headofled, 2017-06-28:
Ok, but what specific file do I download? Do I download the one called "Arducam"? Or is there more?