ArduCAM/Arduino #231: Switch off automatic AWB, ABC and all internal DSP

pranav656, 2017-06-28:
I am currently using the OV2640 for a research project and I would like to control the exposure and switch off all the internal Digital Signal Processing on the sensor. I have achieved exposure control but I would like to switch off the AWB(Automatic White Balancing), ABC(Automatic Black Calibration) and any other on-board Digital Signal Processing that might be happening. To achieve this, I removed the myCAM.OV2640_set_Light_Mode(), myCAM.OV2640_set_Color_Saturation(), myCAM.OV2640_set_Special_effects() and any other functions that modify the image in the code that was provided for OV2640. Is this enough or do I need to do anything else?

Regards,
Pranav

cheribu, 2018-01-06:
I am in the exact same situation.  I have been unsuccessful.

Did anyone respond to your post?  Were you able to achieve your goal?

Thanks,  Cheri

supprot, 2018-01-09:
@cheribu
Hi,
Please try to close the Bit[3] AWB，Bit[2] AWB_GAIN in 0xC3 register Bank=0.