ArduCAM/Arduino #232: OV5642 SPI interface error

e-coucou, 2017-06-30:
Hi,
I received an OV5642 cam, connect to Photon. I check connection and code (SPI ok, CS pin ok ...) but I always receive "SPI interface error with code 255"
Is there an issue with the cam ? no led ? How can I check if the cam is ok ?
Thanks
Eric

UCTRONICS, 2017-07-02:
@e-coucou 
Hi,
Do you use Mini 5MP plus camera ?
If so,the spi error has nothing to do with the camera. 
We suggest you check the spi interface firstly.

Let us know if you need more help.
Regards,
ArduCAM support team.

e-coucou, 2017-07-02:
Thanks for your support.
Yes it's a Mini 5MP Plus Camera. I check the spi interface of the Photon (Particle) with a TFT ST3775 and it works !
I wonder if my cam is alive ? and how to check ?
Regards,
Eric

e-coucou, 2017-07-04:
There is only a wiring issue !!!

closed 2017-07-04 by e-coucou