ArduCAM/Arduino #233: Connecting Nodemcu with ov7076 camera

neerajpal, 2017-07-05:
I am trying to connect ov7076 camera with nodemcu but there are no pin diagram available for the same. As nodemcu have storage equal to 4MBytes and memory 128KBytes. Is there any possibility to use it?

[AI thinker esp8266mod ](https://bennthomsen.files.wordpress.com/2015/12/nodemcu_pinout_700-2.png)

[Omnivision Ov7076](http://i.ebayimg.com/00/s/NTAwWDUwMA==/z/fh0AAOxyUylTVi8n/$_3.JPG?set_id=2)

UCTRONICS, 2017-07-06:
@neerajpal 
Hi,
Our OV7670 camera is used with our ArduCAM shield board.
You can not connect it with modemcu.

Let us know if you nee more help.
Regards,
ArduCAM supprt team.

neerajpal, 2023-09-02:
That's bullshitt, i just did