ArduCAM/Arduino #235: Picture Capture at the press of a button

headofled, 2017-07-10:
My Setup:
-Arduino UNO R3
-4 Arducam OV5642 5MP Plus cameras
-Arducam Multi-camera adapter.

I managed to get the panoramic cam working, but it requires Arducam host to take pictures. Is there a way to do this with a push of a button? Or when a certain if statement is true?

I'm currently running "Arducam_Mini_5MP_Plus_4CAM_Capture2SD"

Any suggestions will be appreciated.

supprot, 2017-07-12:
@headofled 
Hi,
Of course,  you can control the camera take picture by a button.
You can refer to this below code:
keyState= digitalRead(KEY);
if(!keyState)  {
  isCaptureFlag = true;
  while(!digitalRead(KEY));  
}
if(isCaptureFlag){
  myCAM.flush_fifo();
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("start capture."));
  total_time = millis();
  flash_time =millis();
  while ( !myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
  length = myCAM.read_fifo_length();
  if( length < 0x3FFFFF){
    myCAM.flush_fifo();
    myCAM.clear_fifo_flag();
    //Start capture
    myCAM.start_capture();
    while ( !myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
    Serial.println(F("CAM Capture Done."));
    total_time = millis() - total_time;
  }else{
    Serial.println(F("CAM Capture Done."));
    total_time = millis() - total_time;
  }
  total_time = millis();
  read_fifo_burst();
  total_time = millis() - total_time;
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  isCaptureFlag = false;
}
}

Let us know if you need more help.
Regards,
ArduCAM support team.

headofled, 2017-07-12:
@supprot 
I implemented the code you suggested, but I'm getting errors that read:

Arduino: 1.8.3 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Users\Herlihyi\Documents\Arduino\Camera_with_switch\Camera_with_switch.ino: In function 'void myCAMSaveToSDFile(ArduCAM)':

Camera_with_switch:181: error: 'keyState' was not declared in this scope

 keyState=digitalRead(KEY);

 ^

Camera_with_switch:181: error: 'KEY' was not declared in this scope

 keyState=digitalRead(KEY);

                      ^

Camera_with_switch:182: error: 'keystate' was not declared in this scope

 if(!keystate)

     ^

Camera_with_switch:184: error: 'isCaptureFlag' was not declared in this scope

   isCaptureFlag = true;

   ^

Camera_with_switch:187: error: 'isCaptureFlag' was not declared in this scope

 if(isCaptureFlag)

    ^

Camera_with_switch:196: error: 'total_time' was not declared in this scope

 total_time = millis();

 ^

Camera_with_switch:197: error: 'flash_time' was not declared in this scope

 flash_time = millis();

 ^

Camera_with_switch:205: error: 'myCam' was not declared in this scope

   myCam.flush_fifo();

   ^

Camera_with_switch:211: error: 'total' was not declared in this scope

   total_time = millis() - total time;

                           ^

Camera_with_switch:219: error: 'read_fifo_burst' was not declared in this scope

 read_fifo_burst();

                 ^

Camera_with_switch:222: error: 'myCam' was not declared in this scope

 myCam.clear_fifo_flag();

 ^

C:\Users\Herlihyi\Documents\Arduino\Camera_with_switch\Camera_with_switch.ino: At global scope:

Camera_with_switch:226: error: expected unqualified-id before 'if'

 if (length >= MAX_FIFO_SIZE) //8M

 ^

exit status 1
'keyState' was not declared in this scope

I'm assuming that maybe I didn't put the lines in the right place?
If you want, I will copy and paste the code here.

supprot, 2017-07-13:
@headofled 
Hi,
The code on the post just wants you to refer to the button  to trigger taking picture. If you want to get the complete code, please refer to the below link:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_short_movie_clip/ArduCAM_Mini_5MP_Plus_short_movie_clip.ino

Let us know if you need more help.
Regards,
ArduCAM support team.

headofled, 2017-07-20:
@supprot 
Ok, but where do I connect the leads of the switch?

headofled, 2017-07-20:
@supprot 
Also, (2 Things)
1.I tried to use the code found here:
https://github.com/sumotoy/ArduCAM/blob/master/examples/mini/ArduCAM_Mini_5MP_4Cams_TimeElapse2SD_LowPower/ArduCAM_Mini_5MP_4Cams_TimeElapse2SD_LowPower.ino
but when I run it in ArduCam host, the serial monitor keeps saying that OV5642 is not detected despite the fact that I clearly have them plugged into the adapter board.
 
2. I uploaded the code that you gave me and it seems to work, however when I try to view the pictures on the SD card, I get a message that says that the file is empty. 

Solutions/suggestions would be appreciated.

supprot, 2017-07-22:
@headofled 
Hi, 
Sorry for reply you so late.
About the switch, we defined the KEY to GPIO2, In our adapter board,
you can use the button directly.

For SD card,please try to format the sd card in fat32.

Please try it again and let us know if you need more help.

Regards,
ArduCAM support team.

headofled, 2017-07-23:
@supprot 
Where exactly is the GPIO2 on the ArduCam 4 Cameras Adapter Board?
And can you send me a link to fat32?

supprot, 2017-07-23:
@headofled 
Hi,
Please refer to this picture we attached for you
[image]

headofled, 2017-07-27:
@supprot 
So I got the code to work. The video saves to the SD card. 
But for some reason, the video is refusing to playback. I get this error:
[image]
Is there something I should install in order to playback the videos?

headofled, 2017-07-27:
@supprot 
Is there a way to tweak the code you gave me to take photos instead of videos?

UCTRONICS, 2017-07-27:
@headofled 
Hi,
Please use the The storm video to try.

headofled, 2017-08-01:
@UCTRONICS 
Storm video?

UCTRONICS, 2017-08-02:
@ headofled
Hi,
If you want to take photos and save it to your SD card, please refer to this link：

https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_Multi_Capture2SD/ArduCAM_Mini_5MP_Plus_Multi_Capture2SD.ino

Regards,

headofled, 2017-08-03:
@UCTRONICS 
The code works, but it's only taking pictures on 1 camera. I need it to take pictures with all 4 simultaneously.

headofled, 2017-08-03:
@UCTRONICS 
Also, I need it to take pictures with the press of a button (see pictures above), rather than taking series of pictures at timed intervals

UCTRONICS, 2017-08-04:
@headofled
Hi,
If you want to take picture with 4 cameras, you can refer to this demo
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_4CAM_Capture2SD/ArduCAM_Mini_4CAM_Capture2SD.ino

https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_4CAM_Capture2SD/ArduCAM_Mini_5MP_Plus_4CAM_Capture2SD.ino

But you should add button driver by yourself.About button driver, just refer to codes 
we attached for you before.

Let us know if you need more help.
Regards.

headofled, 2017-08-07:
@UCTRONICS 
So I changed around the code, mainly adding: 
#define KEY 2
pinMode(KEY, INPUT);

as well as changing one of the void loops to:
void loop() {
  if(!digitalRead(KEY)){    
  if(CAM1_EXIST)
    myCAMSaveToSDFile(myCAM1);
  if(CAM2_EXIST)
    myCAMSaveToSDFile(myCAM2);
  if(CAM3_EXIST)
    myCAMSaveToSDFile(myCAM3);
  if(CAM4_EXIST)
    myCAMSaveToSDFile(myCAM4);
  delay(5000);
  }

However, and I have no idea if this has anything to do with the code, or the cameras, but the pictures are coming out corrupted, or an improper file, or are not saving entirely. What should I do?

UCTRONICS, 2017-08-08:
@headofled
Hi,
Please try this code we attached for you.
The "pinMode(KEY, INPUT)"; should be written in the void setup(){}.

#define KEY 2
bool keyState;
void setup()
{
.......
......
  pinMode(KEY, INPUT);
}
void loop()
{
  keyState = digitalRead(KEY);
  if (!keyState)  {
    while (!digitalRead(KEY)); //waitting the key release
    if (CAM1_EXIST)
      myCAMSaveToSDFile(myCAM1);
    if (CAM2_EXIST)
      myCAMSaveToSDFile(myCAM2);
    if (CAM3_EXIST)
      myCAMSaveToSDFile(myCAM3);
    if (CAM4_EXIST)
      myCAMSaveToSDFile(myCAM4);
  }
}

Seaatm, 2019-07-16:
Hello, I want to know If you can take a picture and save it inside your computer instead of an SD card

UCTRONICS, 2019-07-17:
@Seaatm 
Yes, if you want to take a picture and save it inside computer, you should use our Host V2 application.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows
[image]

Seaatm, 2019-07-18:
@UCTRONICS I tried and it worked, But is there a way to save the image inside your computer without using the Host V2 app?

UCTRONICS, 2019-07-19:
@Seaatm 
The Arduino UNO board communicates with the PC only through the serial port. If you don't use our host v2 gui, you can use other application which supports communicate through the serial port.