ArduCAM/Arduino #236: "Save Image" default setting

tonehm, 2017-07-11:
Hello, I am using the second generation OV2640 with built-in shield. I want it take pictures and save them on the computer for analysis as one step of a bigger program. I want the picture to be captured and saved automatically. Currently, I have only been able to take pictures with e.g. the V2 Shield code called "ArduCAM_Shield_Video_Streaming" and the host app "ArduCAM_host_v1.0" by clicking "Capture" in the window on the computerand then manually click on "Save Image" and select a path to the folder it should be saved at. Is there a way to implement an automatic default setting to save the images with a corresponding path as a part of the code in the libraries that are posted here? (I'm guessing in the form of some lines in the "ArduCAM_Shield_Video_Streaming" script if that's the one I'm using).

Thanks for any help!

supprot, 2017-07-12:
@tonehm 
Hi,
maybe our Arducam_host_V2 will help.
You can choose Stream mode and choose SaveImage .
In the file address bar, you can set the path where the picture will be saved.

Let us know if you need more help.
Regards,
ArduCAM support team.

tonehm, 2017-07-12:
Thank you for your reply. However, I don't think my question was answered. I have already discovered how I can choose to save an image to a path. But, this must be done manually for each time the code is uploaded to the Arduino. I am looking for and automatic saving function as a part of the code. I have written to you in some other threads as well, but as I have not got an answer to those I can write al the functions I want it to perform here: Ideally my code would in the following order perform:

1. **Change the resolution** (other than the current default setting)
1. **Automatic picture capture** a bit after the program is loaded to the Arduino and the port is opened (the image taken is a step of a bigger procedure and will further be analyzed in another software). So, I want the picture to be taken at a specific time, but automatically!
2. **Automatic saving of this image** to a certain path. Hence, the functions I can manually do in the host window - I want them to be automatic

Hope I made it a bit clearer now what i wish to do.

supprot, 2017-07-19:
@tonehm 
Hi,

Sorry for reply you so late,
Our Host_V2 can't achieve automatic saving of this image. You should develop the host by yourself for your own project.

Let us know if you need more help.
Regards,
ArduCAM support team.

tonehm, 2017-07-21:
Thanks for reply. I have only limited experience with this coding myself;
do you know of an example code other places that can be used?

atiqamuhammad, 2017-12-20:
@tonehm
Hi, i have exactly same problem. I want to take the picture automatically without using host app.. did u find any solution? If u did please email me.
atiqamuhammad96@gmail.com

93450101, 2018-07-06:
@tonehm can you find the way?? @supprot do any thing for me!!!!!!!!!!  I want to collect data of picture arducam mini 2 mp and display image by data i recived .but no body answer me 😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡😡

harmitbadyal, 2019-03-24:
@93450101 @supprot @atiqamuhammad @tonehm have you been able to resolve this issue? I am facing the same issue, I can't automatically save the pictures I take on my computer without using the host app.

UCTRONICS, 2019-03-25:
@93450101 @harmitbadyal 
Hi, sorry for reply you late. Arducam support team can help you collect the image data and stored it into the SD card. 
About the display, different users have different ways.  The Host V2 is just our test application.
If you want to display image, you can use the Host V2 to test if you can get the right image data.

spencerkraisler, 2019-12-05:
@UCTRONICS Is there a way I can access the code of the host app?