ArduCAM/Arduino #238: Broken and shifted images

goetzmoritz, 2017-07-18:
Sometimes it may happen that an image that is taken by the arducam mini 5MP with OV5640 that the actual image is shifted and or color-corrupted. The code I use is from the examples-library for image-capturing. I only altered the wifi-credentials.
Is there any way to avoid that?

[image]

supprot, 2017-07-19:
@goetzmoritz 
Hi,

Please send us the code your are using and we will check it for you.

Let us know if you need more help.
Regards,
ARduCAM support team.

goetzmoritz, 2017-07-25:
I use the basic code that's provided in the samples. But anyways, I think key to this issue is a highly stable power source.

closed 2017-07-25 by goetzmoritz