ArduCAM/Arduino #239: Issue with 4-camera adapter board v2

jialinshi, 2017-07-20:
I'm recently purchased the 4-camera adapter board v2 and 4 ov5647 cameras. I am able to run 2 cameras with the sample code, however running 3 or more cameras gives the error:

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

I am using raspberry pi 3.

ArduCAM, 2017-07-21:
The multi-camera adapter board and pi camera interface is very sensitive to signal integrity.
So please try to connect only one camera to each channel to make sure all the ports are working with single camera.
Then use short FPC cables to the 4 cameras.

jialinshi, 2017-07-21:
I have confirmed that all ports work and they work with any iteration of 2 cameras. Just adding 3+ cameras to the board gives the error.

thoste5, 2017-08-12:
I am getting the same error, but I can't even connect 2 cameras. 1 camera works fine in any of the 4 sockets, but connecting a second anywhere else causes this error. I've tried with both cameras connected individually, and both work fine. The power supply is an official 2.5A one, and I'm using relatively short cables. Are there any other configuration options or tests I could try?

UCTRONICS, 2017-08-13:
@thoste5
Hi,
Please refer to this link to konw how to solve the problems.
http://www.arducam.com/knowledge-base/raspberry-pi-multi-camera-adatper-board/

Regards.

snosenzo, 2018-04-03:
Hey, I had this problem as well. 
The solutions above don't make much sense to me and didn't really work. I found a workaround in the meantime even though it might not be advisable to unplug and replug the cameras while the pi is on. Here's what I've been doing:
To test to see that the cameras are working I use
`raspistill -v`
if that returns the mmal_...  error then I do the following:

1. I unplug my raspberry pi from its power source (to turn it off)
2. Feed one of the camera's ribbon cables into the pi (unplugging it from the arducam multi-camera adaptor)
3. Then I plug the pi back into its power source to turn back on
4. After it's turned back on I run the `raspistill -v` command in the terminal to see that a single camera works if it doesn't return the error. If it does I shut the pi off again and unplug and replug the camera ribbon cable.
5. If it did work then I unplug the single camera from the pi, Plug it back into the arducam adaptor, and then plug the adaptor back into the pi (while the raspberry pi is on). I'm sure that this is not advisable to do. But I've done this about a dozen times and nothing has happened.
6. Then I run `raspistill -v` again and it still works (doesn't end with an error
7. Then I run a python script I wrote to setup the gpio pins and capture an image (much like the example)

This surely isn't a great solution, but it's worked for us. It's really annoying to have to do this every time the pi turns off because it looks like it essentially has to boot up with only one camera plugged into it, and then be changed the to multi-camera adaptor for it to work.

Hope that helps, we haven't really encountered a better solution yet, but this is working so far.

ArduCAM, 2018-04-03:
I'm sorry to hear that you have such problem, would you please send me some clear hardware setup photos, so I can isolate the problem, if you can send me your email address to admin@arducam.com, it will be much easier to discuss.

mkeyno, 2019-12-16:
@ArduCAM 
I' have the same problem, I've tried to follow those solutions on this ticket, and this example
https://www.arducam.com/multi-camera-adapter-module-raspberry-pi/
but no success  
```
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

```
I bought these items from 
https://www.iot-store.com.au/products/ov5647-camera-board-with-m12-mount-ls-40180-fisheye-lens-for-raspberry-pi
https://www.iot-store.com.au/collections/arducam/products/multi-camera-adapter-module-fully-compatible-for-official-raspberry-pi-board

any help really appreciated

UCTRONICS, 2019-12-16:
@mkeyno 
Have you run **vcgemcmd get_camera** to check the camera?

UCTRONICS, 2019-12-16:
@mkeyno 
Hi,
Have you run **vcgencmd get_camera** to check the camera ?

mkeyno, 2019-12-16:
@UCTRONICS 
it replys ``` supported=1 detected=0```

UCTRONICS, 2019-12-16:
@mkeyno 
The detected=0 means the system has not detected the camera, Please attach me your hardware setup.
I guess your hardware is old version. For the new official raspberryPi  firmware v9.9 or above. It does not compatible with our old board. Maybe you should use our new version board or use the old version raspberrypi firmware. You can download it from http://downloads.raspberrypi.org/raspbian/images/

mkeyno, 2019-12-16:
@UCTRONICS 
I've just bought my RPi and updated with last possible repository 

Kernel		: Linux 4.19.75-v7l+ (armv7l)
Version		: #1270 SMP Tue Sep 24 18:51:41 BST 2019
C Library		: GNU C Library / (Debian GLIBC 2.28-10+rpi1) 2.28
Distribution		: Raspbian GNU/Linux 10 (buster)
-Current Session-
Computer Name		: raspberrypi
User Name		: pi ()
Language		: en_NZ.UTF-8 (en_NZ.UTF-8)
Home Directory		: /home/pi
-Misc-
Uptime		: 1 hour 22 minutes
Load Average		: 0.29, 0.31, 0.24
Available entropy in /dev/random		: 3587 bits (healthy)

UCTRONICS, 2019-12-16:
@mkeyno 
Your FPC line is reverse.
[image]

mkeyno, 2019-12-16:
@UCTRONICS 
I feel like a dummy , thanks anyway , its work now

hansangseok, 2024-08-13:
hi, i have a question about 4-camera adapter board v2.2 
Can i use this in ubuntu 20.04.6 Not Raspbian???