ArduCAM/Arduino #240: Native ESP32 interface

kevinkessler, 2017-07-21:
I'm trying to interface an ArduCam 5MP OV5642 mini to an ESP32 without any Arduino libraries, so I'm basically taking the existing libraries, replacing the i2c and spi methods, and just putting everything in a flat C file.  I have the i2c and spi interfaces looking like they work properly; the values in the program match what shows up on my logic analyzer, they read the right chip IDs,  I set the JPEG size to a higher resolution the fifo_length gets bigger, etc.  The problem is, no matter how I read the FIFO buffer (BURST or SINGLE), I do not get something that looks like a JPEG file.

The OV5642 datasheet that I have does not show a simple register that says "Output format = JPEG" or anything like that, and the sensor_reg arrays in ov5642_regs.h are very confusing, with the same registers being set to different values in the same array (OV5642_QVGA_Preview is especially crazy).

What are the minimum i2c register addresses and values I need to set to force the ArduCam to dump out a JPEG? I don't care how deformed the output picture would be, I just would like to see the ArduCam put out something that starts with 0xFF 0xD8 and ends with 0xFF 0xD9.

supprot, 2017-07-22:
We have ready to use ESP32 library and examples.
You'd better to test our example before porting your own flat C code.
Please check: https://github.com/ArduCAM/ArduCAM_ESP32S_UNO

zexanana, 2018-08-05:
@kevinkessler did you ever get around to solving this issue and finishing the flat c library for the OV5642?

frka1204, 2018-08-11:
I'm also trying to interface ArduCam 5MP OV5642 with ESP32 and I have the same problem. The first bytes in the jpeg are often correct (= FFD8) but I never encounter FFD9.

ArduCAM, 2018-08-13:
You can try the camera with standard Arduino UNO or our arducam ESP32 UNO board, the examples are tested on these board.

frka1204, 2018-08-21:
It seems I found a solution to my problem. I added an extra time delay in the code where register values are written to the image sensor:

wrSensorRegs16_8(OV5642_QVGA_Preview);
vTaskDelay(100 / portTICK_RATE_MS);
vTaskDelay(100 / portTICK_RATE_MS); // Extra time delay added here!
wrSensorRegs16_8(OV5642_JPEG_Capture_QSXGA);

misghna, 2020-01-23:
@frka1204  I'm also trying to interface ArduCam 5MP OV5642 with ESP32 , do you have the sample code by any chance. Thanks,

frka1204, 2020-01-24:
Hello

I attach the files that I'm using. You probably have to make some modifications...
Here's an example how to fetch the image data:

uint8_t * imageBuf = (uint8_t *) heap_caps_malloc(65536 * 2, SPI_MALLOC_CAP);
if (!imageBuf)
printf("Allocation failed");
getImageData(imageBuf);
releaseSPIMaster();

/Fredrik

misghna, 2020-01-24:
test
[Seeed-OV2640.pdf](https://github.com/ArduCAM/Arduino/files/4109234/Seeed-OV2640.pdf)

misghna, 2020-01-24:
@frka1204  Thanks for your prompt replay, but I don't see any attachments in your message.