ArduCAM/Arduino #241: ESP8266 with arduino board

goetzmoritz, 2017-07-25:
Hi,

I'd like to use the ESP8266 board but not with the board-repository provided by arducam as many other libraries are not compatible.
How is it possible to use an ESP8266/NodeMCU with Arducam and the stock ESP8266-Arduino-Core https://github.com/esp8266/Arduino?

thanks!

goetzmoritz, 2017-07-25:
Just FYI: Arducam is not compatible with https://gitlab.com/BlackEdder/painlessMesh

I get error messages like that:

Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_pair.h:59:0,

                 from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\utility:70,

                 from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\tuple:38,

                 from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:55,

                 from \Documents\Arduino\libraries\painlessMesh-master-75d2c551c41c62a8d96630d0cd8c186b6da2cb48\src/painlessScheduler.h:203,

                 from \Documents\Arduino\libraries\painlessMesh-master-75d2c551c41c62a8d96630d0cd8c186b6da2cb48\src/painlessMesh.h:10,

                 from \AppData\Local\Temp\arduino_modified_sketch_161281\startHere.ino:16:

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\move.h:166:28: error: macro "swap" requires 3 arguments, but only 2 given

     swap(_Tp& __a, _Tp& __b)

                            ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\move.h:185:42: error: macro "swap" requires 3 arguments, but only 2 given

     swap(_Tp (&__a)[_Nm], _Tp (&__b)[_Nm])

                                          ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\move.h:187:38: error: macro "swap" requires 3 arguments, but only 2 given

     noexcept(noexcept(swap(*__a, *__b)))

                                      ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\move.h:191:25: error: macro "swap" requires 3 arguments, but only 2 given

  swap(__a[__n], __b[__n]);

                         ^

In file included from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\utility:70:0,

                 from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\tuple:38,

                 from \appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:55,

                 from \Documents\Arduino\libraries\painlessMesh-master-75d2c551c41c62a8d96630d0cd8c186b6da2cb48\src/painlessScheduler.h:203,

                 from \Documents\Arduino\libraries\painlessMesh-master-75d2c551c41c62a8d96630d0cd8c186b6da2cb48\src/painlessMesh.h:10,

                 from \AppData\Local\Temp\arduino_modified_sketch_161281\startHere.ino:16:

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_pair.h:194:21: error: macro "swap" requires 3 arguments, but only 1 given

       swap(pair& __p)

                     ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_pair.h:195:46: error: macro "swap" requires 3 arguments, but only 2 given

       noexcept(noexcept(swap(first, __p.first))

                                              ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_pair.h:196:44: error: macro "swap" requires 3 arguments, but only 2 given

[... 518 lines cut ...]
                 from \Documents\Arduino\libraries\painlessMesh-master-75d2c551c41c62a8d96630d0cd8c186b6da2cb48\src/painlessMesh.h:15,

                 from \AppData\Local\Temp\arduino_modified_sketch_161281\startHere.ino:16:

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\shared_ptr.h:431:52: error: macro "swap" requires 3 arguments, but only 2 given

     swap(shared_ptr<_Tp>& __a, shared_ptr<_Tp>& __b) noexcept

                                                    ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\shared_ptr.h:432:19: error: macro "swap" requires 3 arguments, but only 1 given

     { __a.swap(__b); }

                   ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\shared_ptr.h:517:48: error: macro "swap" requires 3 arguments, but only 2 given

     swap(weak_ptr<_Tp>& __a, weak_ptr<_Tp>& __b) noexcept

                                                ^

\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\shared_ptr.h:518:19: error: macro "swap" requires 3 arguments, but only 1 given

     { __a.swap(__b); }

goetzmoritz, 2017-07-25:
It is possible to #undef swap but than IP2STR pops up and it is not possible to #undef that away. Any suggestions?

supprot, 2017-07-26:
@goetzmoritz 
Hi,

Our ArdduCAM ESp8266 can be compiled successfully on NodeMCU 1.0 platform. But If you want to 
support  the stock ESP8266-Arduino-Core https://github.com/esp8266/Arduino
You should modify the underlying board level file which will be very difficult.
[image]

Let us know if you need more help.
Regards.

closed 2017-08-11 by goetzmoritz